FreeCAD/FreeCAD #10527: Merge the Help module

yorikvanhavre, 2023-09-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The current Help module, currently installable as an add-on, is now the only way to access the FreeCAD documentation from within FreeCAD, as the older way, which relied on enormous, unmanageable Qt help files, has been removed.

When using any of the help features (Help menu, "what's this" button), if the Help module is not installed, you get a "please install the Help add-on" message.

So I think we should add the Help module to the built-in modules, so it is installed by default. The Help module is extremely lightweight, it is composed of basically one Python file plus a couple of additional things (preferences page, etc)

A couple of points that should be fixed before doing so:

* Remove the dependency on the `markdown` python module, or at least still work nicely without it
* Maybe open the wiki by default, instead of the markdown version?

### Full version info

```shell
0.22dev
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-09-23:
#### Note to merge dev, please add `Fixes #10512` in to the PR that fixes this ticket (10527) TIA!

yorikvanhavre, 2023-10-03:
The current version of the Help addon at https://github.com/FreeCAD/FreeCAD-Help is now in principle ready for merging.

* [x] The wiki is now the default source
* [x] Use of markdown conversion modules (markdown or pandoc) is now optional, and only used if the user switches from the wiki to markdown sources. If none of these modules is present, an ugly but functional built-in rendering is still performed, so FreeCAD will not depend on any additional module
* [x] Translation system is in place

Remaining to do: 
* [ ] cleanup and remove addonManager files
* [ ] update the wiki

PR #11008 (2023-10-16): Merging the Help module
Merging the help module from https://github.com/FreeCAD/FreeCAD-Help 

fixes #10527
fixes #10512
closed 2023-10-16 by chennes