FreeCAD/FreeCAD #10528: [Problem] Access violation with PD Mirror

FEA-eng, 2023-09-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The "Access violation" error keeps occuring when trying to mirror the sweep feature in a particular model. It doesn't always happen when modeling this part from scratch in the same way. Suspected coincidence issue. It happens in FreeCAD 0.21 and 0.22, on different computers.

Forum discussion: https://forum.freecad.org/viewtopic.php?t=80888

File: 
[Access violation 2.zip](https://github.com/FreeCAD/FreeCAD/files/12514587/Access.violation.2.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34137 (Git)
Build type: Release
Branch: master
Hash: 7ca834acc58841dcf2d66d186db818c8f35f359d
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2023-09-06:
stacktrace
```Err: Program received signal SIGSEGV, Segmentation fault.
Err: #0  /lib/x86_64-linux-gnu/libc.so.6(+0x3bfd0) [0x7fb44e25afd0]
Err: #1  0x7fb3f80c7628 in Geom2d_Curve::Value(double) const from /lib/x86_64-linux-gnu/libTKG2d.so.7+0x18
Err: #2  0x7fb3697fbb36 in BOPAlgo_PaveFiller::FillPaves(int, int, int, NCollection_List<opencascade::handle<BOPDS_PaveBlock> > const&, opencascade::handle<BOPDS_PaveBlock> const&) from /lib/x86_64-linux-gnu/libTKBO.so.7+0x186
Err: #3  0x7fb3697fd543 in BOPAlgo_PaveFiller::ProcessDE(Message_ProgressRange const&) from /lib/x86_64-linux-gnu/libTKBO.so.7+0x973
Err: #4  0x7fb3697b4c4c in BOPAlgo_PaveFiller::PerformInternal(Message_ProgressRange const&) from /lib/x86_64-linux-gnu/libTKBO.so.7+0x144c
Err: #5  0x7fb3697b5bec in BOPAlgo_PaveFiller::Perform(Message_ProgressRange const&) from /lib/x86_64-linux-gnu/libTKBO.so.7+0x4c
Err: #6  0x7fb36971fb0e in BRepAlgoAPI_BuilderAlgo::IntersectShapes(NCollection_List<TopoDS_Shape> const&, Message_ProgressRange const&) from /lib/x86_64-linux-gnu/libTKBO.so.7+0x14e
Err: #7  0x7fb36971db79 in BRepAlgoAPI_BooleanOperation::Build(Message_ProgressRange const&) from /lib/x86_64-linux-gnu/libTKBO.so.7+0x7d9
Err: #8  0x7fb367d44672 in PartDesign::Transformed::execute() from /home/user/Programs/FreeCADmaster/build/Mod/PartDesign/_PartDesign.so+0xe12
Err: #9  0x7fb4508b27c2 in App::DocumentObject::recompute() from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADApp.so+0x92
Err: #10  0x7fb36ba9326b in Part::Feature::recompute() from /home/user/Programs/FreeCADmaster/build/Mod/Part/Part.so+0xb
Err: #11  0x7fb45085a09c in App::Document::_recomputeFeature(App::DocumentObject*) from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADApp.so+0x28c
Err: #12  0x7fb450861711 in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADApp.so+0x571
Err: #13  0x7fb450916194 in App::DocumentPy::recompute(_object*) from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADApp.so+0x2c4
Err: #14  0x7fb4509162a3 in App::DocumentPy::staticCallback_recompute(_object*, _object*) from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADApp.so+0x23
Err: #15  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x1c9ba1) [0x7fb44fbc9ba1]
Err: #16  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x7d) [0x7fb44fb7be8d]
Err: #17  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x4a63) [0x7fb44fb09c63]
Err: #18  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x3ccd3c) [0x7fb44fdccd3c]
Err: #19  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(PyEval_EvalCode+0xc1) [0x7fb44fc68891]
Err: #20  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x2b7359) [0x7fb44fcb7359]
Err: #21  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(PyRun_StringFlags+0x6b) [0x7fb44fcb850b]
Err: #22  0x7fb45032dba8 in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADBase.so+0x58
Err: #23  0x7fb451412f60 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADGui.so+0x80
Err: #24  0x7fb4514130bc in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADGui.so+0xcc
Err: #25  0x7fb4514131c2 in Gui::Command::updateActive() from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADGui.so+0x32
Err: #26  0x7fb367514c85 in finishFeature(Gui::Command const*, App::DocumentObject*, App::DocumentObject*, bool, bool) from /home/user/Programs/FreeCADmaster/build/Mod/PartDesign/PartDesignGui.so+0x235
Err: #27  /home/user/Programs/FreeCADmaster/build/Mod/PartDesign/PartDesignGui.so(+0x117289) [0x7fb367517289]
Err: #28  /home/user/Programs/FreeCADmaster/build/Mod/PartDesign/PartDesignGui.so(+0x119c03) [0x7fb367519c03]
Err: #29  0x7fb36751fc97 in prepareTransformed(PartDesign::Body*, Gui::Command*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::function<void (App::DocumentObject*, std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> >)>) from /home/user/Programs/FreeCADmaster/build/Mod/PartDesign/PartDesignGui.so+0x357
Err: #30  0x7fb3675203c5 in CmdPartDesignMirrored::activated(int) from /home/user/Programs/FreeCADmaster/build/Mod/PartDesign/PartDesignGui.so+0xc5
Err: #31  0x7fb45141b164 in Gui::Command::_invoke(int, bool) from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADGui.so+0x264
Err: #32  0x7fb45141b5c3 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADGui.so+0x133
Err: #33  /lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2e8f4f) [0x7fb44e6e8f4f]
Err: #34  0x7fb44f35c782 in QAction::triggered(bool) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x42
Err: #35  0x7fb44f35f3ab in QAction::activate(QAction::ActionEvent) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xab
Err: #36  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x256e8a) [0x7fb44f456e8a]
Err: #37  0x7fb44f456fd7 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xf7
Err: #38  0x7fb44f54faaa in QToolButton::mouseReleaseEvent(QMouseEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xa
[... cut ...]
Err: #62  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x85) [0x7fb44e246285]
Err: #63  /home/user/Programs/FreeCADmaster/build/bin/FreeCAD(+0x9d11) [0x558c10298d11]

```

version
```OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34197 (Git)
Build type: Release
Branch: master
Hash: 80a54d51c1c212477a116f70e6bdbeb70e7628a7
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * QuickMeasure 2022.10.28
```

FEA-eng, 2025-09-14:
I can't reproduce it at the moment. I wonder if something fixed it.

FEA-eng, 2025-12-14:
> [@FEA-eng](https://github.com/FEA-eng) is this still not reproducible for you ?

Yeah, I can't reproduce it now. But I'm always skeptical in such cases when there's no known fix.

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20251209 (Git shallow)
Build date: 2025/12/09 13:53:17
Build type: Release
Branch: grafted,grafted
Hash: 5ad6a1ba58c665c23f1d217c23e74379d8c1a5a4
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

luzpaz, 2025-12-15:
@FEA-eng we can close for now and if it crops back up we can re-open