FreeCAD/FreeCAD #10532: [Problem] Part Design, thickness feature doesn't work with specific B-splines pockets

benj5378, 2023-09-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Depending on how the B-spline is located and used to pocket a solid, the thickness feature doesn't work.

[M12.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/12515118/M12.FCStd.zip)

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34125 (Git) Snap 761
Build type: Release
Branch: master
Hash: ea116dc0332ccfc5797f248b04492a16f421b787
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * Help 1.0.3
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2023-09-05:
This is not b-spline related, it is a general issue.

Duplicate #6055 and  #5829

benj5378, 2023-09-05:
Closed as duplicate

closed 2023-09-05 by benj5378