FreeCAD/FreeCAD #10533: [Problem] Part Design, Mirrored feature mirror plane selection doesn't work

benj5378, 2023-09-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Mirrored feature task shows some planes, that can be clicked and selected, such they turn green. I therefore the selection of one of them to become the mirror plane. But this is not the case.

Not sure if this a bug or feature request.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34125 (Git) Snap 761
Build type: Release
Branch: master
Hash: ea116dc0332ccfc5797f248b04492a16f421b787
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * Help 1.0.3
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2023-09-05:
A minimal example file? Guessing, if it is no datum plane, then maybe a b-spline surface, which looks like a plane.

benj5378, 2023-09-05:
> A minimal example file?

No need: 1) make sketch, 2) pad, 3) make mirror feature and see the behaviour of plane selection in 3D view.

> Guessing, if it is no datum plane, then maybe a b-spline surface, which looks like a plane

We are talking about the base planes (XY, YZ, XZ)

grubertechnic, 2023-09-05:
> We are talking about the base planes (XY, YZ, XZ) (but this can be expanded on I guess)

This is a information, which is missing, else it is ambiguous.

But still, works for me in all datum plane directions. If that would not be working in general, there would be already be reported i think. Weird that such a fundamental task does not work.

[image]

my tested version:

```
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34188 (Git)
Build type: Release
Branch: master
Hash: e248e501e57585c84d073e5aff152e1deacb0799
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * QuickMeasure 2022.10.28
```

grubertechnic, 2023-09-05:
Yes. Same. It works (on my compiled master version).

benj5378, 2023-09-06:
> Yes. Same. It works (on my compiled master version).

Are you sure you get me? It's not what your screenshot shows

grubertechnic, 2023-09-06:
> Are you sure you get me?

When you asking again, no, i do not get your point. Maybe a step by step instruction would help. At least yours from before works for me.

aavogt, 2024-01-10:
I also expected the `Gui.Selection.addSelection` to set the `MirrorPlane` attribute. Here is a video which also shows that Pocket cannot be mirrored:

[freecad_mirror_plane_selection.webm](https://github.com/FreeCAD/FreeCAD/assets/93006/aab8e071-9cb9-472c-94d8-28b9ff56fb43)

benj5378, 2024-01-13:
I have thought a bit about this. As base plane selection is used many places throughout FreeCAD, there should propably be a global plane selection implementation.

Pendelf, 2024-07-28:
I understand, you need to choose a plane for reflection, but do this in the 3d view window does not work

benj5378, 2025-01-24:
This is still an issue

Pendelf, 2026-10-05:
Still relevant

OS: Windows 11 build 26300
Architecture: x86_64
Version: 26.3.0dev.20261001 (Git shallow)
Build date: 2026/10/01 12:05:00
Build type: Release
Branch: (HEAD detached at 99c5620c5)
Hash: 99c5620c5f739ca5c308849dda95ac468fec8b79
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 617.14/NVIDIA Corporation/NVIDIA GeForce RTX 5060 Ti/PCIe/SSE2
Installed mods:
  * AIGenFurniture Cabinet Generator 0.2.0
  * AddFC Workbench 3.7.8
  * Curved Shapes 1.0.15
  * Curves workbench 0.6.81
  * DFM 0.1.21
  * Fasteners Workbench 0.5.67
  * FrameForge 0.2.1
  * History 0.2.0
  * Nesting 2026.9.0
  * PieMenu 1.13.0
  * SheetMetal Workbench 0.8.24
  * StructureTools 0.0.1
  * SvgWorkbench 1.0.0.dev15