FreeCAD/FreeCAD #10534: [Problem] Shape binder is relocated upon sketch edit.

benj5378, 2023-09-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a Sketch004 that's Pad into a solid. Another Sketch005 is made on one of the surfaces of this solid and a Shape Binder is made of the sketch and then translated along the Z axis.

Now, if a dimension of Sketch004 is changed, the Shape Binder is relocated to it's original position upon creation. Is this intended behaviour? It doesn't seem to be TNP related, as no objects are reordered/deleted?

Also, why doesn't ShapeBinder have transform property in Data tab, if it can be translated?

[M12.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/12515586/M12.FCStd.zip)

https://github.com/FreeCAD/FreeCAD/assets/12400097/e61ae324-2196-4cbc-a448-3ec1d5c31c49

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34145 (Git) Snap 763
Build type: Release
Branch: master
Hash: 0a8292a850b98d2e66e08e235dfb81b6e7285df1
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * Help 1.0.3
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2023-09-05:
I do not get this far, i can not move with Draft move the sketch.

Error message:
`Sketch005 cannot be modified because its placement is readonly.`

Draft operations are also not made for PartDesign. It have a complete different placement/attachment behavior.

benj5378, 2023-09-05:
> I do not get this far, i can not move with Draft move the sketch.

The ShapeBinder is moved, not the sketch, please try again.

grubertechnic, 2023-09-05:
True, i missed that. But yes, that is expected, You do not need to edit the sketch, that it flip back. You need only to recompute the shapebinder. It has no (default editable) placement. (it has one if you look in the properties, which is hidden for shapebinders, but it strict follows the placement of the target).

For that is the subshapebinder. And yes, that is confusing and there is already a thread with a feature request, which wants to union that. And i also support that.

see

https://forum.freecad.org/viewtopic.php?t=80880
https://forum.freecad.org/viewtopic.php?t=76445

grubertechnic, 2023-09-05:
When i have more time i want to crate a ticket for that, but for that i need more time, because i want to write that out.

benj5378, 2025-01-24:
Closed as this is not an issue for subshapebinders. Seems like issue arrised from confusion between different (sub)shapebinders.

- #13044