FreeCAD/FreeCAD #10536: [Problem] Scaling a Part shape is awkward

WandererFan, 2023-09-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Current situation:
In order to scale a shape, one must load the Draft workbench and create a Clone of the shape.  3 properties in the Clone must then be updated with the scale factor.
There is also a Scale tool in Draft, but it does not perform a simple "scale to 2x" operation.

Desired situation:
Select one or more source shapes, press a Scale button in the Part toolbar.  This would create a "Scaled" DocumentObject whose Shape property contains a scaled version of the source shapes.

### Full version info

```shell
v0.22
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-09-05:
You can also create a Link which has a `Scale` and a `Scale List` property.

luzpaz, 2023-09-09:
> I think I missed something for objects that are already selected when the tool is started.

WandererFan, 2023-09-12:
The current task dialog does not accept negative factors, but they can be set in the property editor afterwards.  I will change the dialog to accept negative numbers from the outset.

I don't know how to make the factors expression aware yet.

wwmayer, 2023-09-13:
A widget can only be bound to a property of a specific scale object. But because the scale objects will be created after clicking the Apply/OK button this isn't possible.

All what would be possible is to implement the edit function of the view provider of an existing scale object so that after its creation the properties can be bound to the widgets. Therefore the scale dialog must be reworked that an existing scale object can be passed and that in this state it will only allow to modify this object and not to create new scale objects.

WandererFan, 2023-09-13:
Task dialog has been updated to allow negative factors.

As werner pointed out, expressions can only be used once the object exists, so can't be applied in the creation dialog.  Once the object exists, you can apply an expression to the Scale object in the PropertyEditor.

closed 2023-09-13 by WandererFan