FreeCAD/FreeCAD #10538: [Problem] No good way to create walls / thin extrusions in Part Design

kadet1090, 2023-09-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently there is no good way to create "Thin" elements in the FreeCAD. Because of that creating wall-like elements in the Part Design is really hard and requires a lot of effort. Possible uses:
- Creating walls
- Creating ribs
- Creating reinforcements

To illustrate this feature in OnShape:
[image]

It'd be nice to implement this in a way that would allow this feature to be easily used in most (all?) features without code duplication, so It probably should be implemented as separate "thicken" feature, that then can be: padded, pocketed, revolved, etc. 

This potential feature should also allow control on:
 - Offset (One way, symmetric, Two Distances)
 - Cap type (Round / Butt)
 - Join type (Round / Tangent)
 
If possible, there should be a way to create "Thicken" feature using for example "Pad" dialog, to simplify user workflow and make it more user friendly.

### Workarounds
- Using SheetMetal add base: https://wiki.freecad.org/SheetMetal_AddBase
- Creating SubShape binder to sketch with filled offset applied - it works, but it is not intended use of binder, and it has a lot of quirks
- Using Part workbench with 2d offset
- Sketch Offset tool will help a lot with many cases when (if) merged, but again it's not correct tool for the job
- Using Arch_Wall ?

### Full version info

```shell
FreeCAD 0.21
```

### Subproject(s) affected?

PartDesign

### Anything else?

 - https://forum.freecad.org/viewtopic.php?t=31624
 - https://youtu.be/c98dkN8yt-4?t=717 - Clough42 video featuring this in Fusion360
 - https://youtu.be/hP4UNtoYGSc - Video introducing that feature to OnShape
 
Technically, this is duplicate of https://github.com/FreeCAD/FreeCAD/issues/6057 but this issue lacks description and is old so It'd be hard to get proper feedback.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kadet1090, 2023-09-04:
For the implementation I'd see that as separate feature, just like I described in the issue. This however would not be as simple as integrating "Offset 2D" feature into PartDesign. This would be the simplest solution, but world is rather cruel and often the easiest possible solution is not the correct one. Why? Because "Offset 2D" has a lot of quirks, is really picky for creating two way offsets and gives no control on for example cap type.

My proposal is to create new utility class for creating offsets. It should take all necessary parameters (Offset Inwards, Offset Outwards, Cap Type, Join Type) and work roughly like this:
1. create offsets using two calls to `BRepOffsetAPI_MakeOffsetShape::performBySimple` or `BRepOffsetAPI_MakeOffsetShape::performByJoin`, creating pair of offseted paths. If one of of offset is basically 0 (`< tolerance()`) then simply original shape should be used.
2. connect pairs of dangling path ends using:
 - line for butt type
 - arc for round type

From my initial tests it seems that creating path offset is less picky than trying to create whole offset in one go using "MakeOffsetShape" API. As probably most offseted paths would consist only of lines and simple arcs, using `performBySimple` should be possible in most cases, which should give more stable results.

This however would work only for geometry that has no holes, implementing it for more sophisticated cases would require more effort, so I propose to split this issue into 3 stages. For the later part of this comment I'd treat input geometry as graph structure (i.e. vertices connected by edges) and use graph related nomenclature.

0. Create utility class, that could create offsets using method described above.
1. Create first stage that would work only with simple shapes - simple paths, and cycles - basically connected graphs with maximum degree <= 2
2. Create second stage that would extract each component of graph, use method from first revision and union all results together
3. Create third revision that would work with graphs that has degree > 3 (e.g. has Y junctions). This could be done via splitting each extracted component into cycles and paths (in the best scenarios as few possibly largest ones) and then proceeding like in stage 2.

Each stage is worth of publishing on its own as feature would be useful even if only stage 1 is implemented, as it already covers a lot of use cases. Having stage 1 stage 2 should also be simple to implement (extracting components from graphs is easy operation). The most trouble would be to implement stage 3, but this is case where other CADs also struggles so.

maxwxyz, 2023-10-20:
The offset for starting the operation would also be great in the normal Pad, Pocket and Hole features!

I have created an issue for the Pad feature "up to face", "to last" and "to first" as how it is currently implemented works not as expected. This would benefit the "end profile" for where you need the ribs to be extruded to (to join the ribs with the shell contour / face curvature):
https://github.com/FreeCAD/FreeCAD/issues/11111

adrianinsaval, 2023-10-20:
the big problem with trying to implement this is that both offset 2D and 3D are crap on occt, very prone to error. If we implement this I think we will have endless complaints of the tool not working, with the actual problem entirely out of our control since it's in the kernel.

kadet1090, 2023-10-20:
OCCT does not have notion of Offfset 2D that is why it works so bad, because offset 2D is actually calculated in 3 dimensions instead of 2, which makes the problem significantly harder. For example OCCT is not able to calculate offset of line because it does not know which normal to pick. offsetting things like faces from my experience works okayish. 

That's why in implementation I think that we have to do some own work and only use some features from OOCT that work well enough.

 >the actual problem entirely out of our control since it's in the kernel.
That is a big problem for FreeCAD in the future, IMHO we should at least think about ability to compile FreeCAD with custom patches to OCCT kernel so we can fix some issues ourselves and maybe even upstream them some day.

adrianinsaval, 2023-10-20:
> > the actual problem entirely out of our control since it's in the kernel.
> 
> That is a big problem for FreeCAD in the future, IMHO we should at least think about ability to compile FreeCAD with custom patches to OCCT kernel so we can fix some issues ourselves and maybe even upstream them some day.

actually all of our official packages are compiled using a patched version already: https://gitlab.com/blobfish/occt

geometric kernel development is a pretty complex matter though, and unless we can really commit to maintaining a full fork it is not sustainable to have big changes compared to upstream. Not to mention that we don't really have people to make such big changes.

I hope the increasing funding we are getting may help us be more active on the occt side in the future, hopefully by submitting patches upstream rather than a full fork. There are many things from occt that are limiting FreeCAD's capabilities.

kadet1090, 2023-10-20:
I also don't think that full fork is feasible, but set of patches for simpler cases that can be rebased to each OCCT release and hopefully someday merged into master would be a great addition.

adrianinsaval, 2023-10-20:
yeah that's what we are doing, although I think blobfish doesn't want to submit his patches upstream due to their contributor agreement BS

maxwxyz, 2024-03-09:
Here is another example from SolidWorks, creating thin features when extruding a line profile:
[image]

FlachyJoe, 2024-03-09:
1. Create a sketch
2. Create a subshapebinder of it
3. Set the Offset parameter
4. Extrude the subshapebinder

[image]

Hi-Angel, 2026-08-17:
> yeah that's what we are doing, although I think blobfish doesn't want to submit his patches upstream due to their contributor agreement BS

For future readers — IIUC, nowadays FreeCAD is no longer using the fork *(last change to the fork was in 2024 but [next FreeCAD release is ready to work with OCCT 8](https://github.com/FreeCAD/FreeCAD/pull/25502))*.

codeotaku, 2026-09-08:
Exploring how to do this in FreeCAD.  Thin extrusions aren't too bad. Working on trimming with shape boundaries for ribs etc. 

https://github.com/user-attachments/assets/9878bf81-897a-4820-b197-5bc767864f9b

codeotaku, 2026-09-08:
Ok, trimming based on (whole) shape intersections is working 

[image]

codeotaku, 2026-09-08:
thin cut/extrude kind of working... just trying to cover off some basics before (attempting) to build this feature

[image]

maxwxyz, 2026-09-09:
Nice, thanks for working on this. Does this work with profiles from above (lines) and from the side, to create e.g. angled ribs on flanges?

codeotaku, 2026-09-09:
Yes, exactly that.  I'm working making a more streamlined UI - and separate tools for ribs vs webs - but the core functionality is there.  

Currently, I have the "web" feature as an extension to Pad/Pocket (but works with edges rather than faces, of course) - and can extend up drafted/curved walls/bowl shapes (which was so difficult to deal with before!)  Cut/Extrude is in the Pocket tool.

Working on the rib feature (probably makes sense as its own tool?) - the core is pretty much the same, but it needs some work on the interface... 

[image]

[image]

maxwxyz, 2026-09-09:
IMO in that case (profile from the side) the dragger could also drag its width and the profile being extended from the line to the adjacent faces automatically.

codeotaku, 2026-09-09:
@maxwxyz  Yes, those options are there too! 

[image]

I'm adding the draggers back in - I agree.

maxwxyz, 2026-09-09:
I'm removing GSoC label if you are implementing this! Thanks!!

codeotaku, 2026-09-09:
Sounds good! Should have this for review this week

freecad-ci-runner, 2026-09-09:
This issue was marked for UI/UX review. If you'd like to talk it over with the group, join the discussion at https://discord.com/channels/870877411049357352/1547273184988897410, or ping `@FreeCAD/design-working-group` here. The discussion is recorded and will be available at https://wg.freecad.org/logs/thread/1547273184988897410.