FreeCAD/FreeCAD #10543: [Problem] Draft array tool does not allow use of expressions during initial creation

atlast-50, 2023-09-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The draft array tool is inconsistent in that it does not allow expressions during initial creation, even though they can be used by modifying the properties of the array after creation.

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.2
[/code]
```

### Subproject(s) affected?

Draft

### Anything else?

No expressions allowed here
[image]

Expressions work fine when modifying properties
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-01-30:
Related forum post:
https://forum.freecad.org/viewtopic.php?t=83660

xorxx, 2024-04-29:
Would be nice to have the function buttons inside the dialogue. Would make it more intuitive to use.