FreeCAD/FreeCAD #10545: [Problem] Exporting an array object gives an incorrect result

atlast-50, 2023-09-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When exporting a part made with the draft array tool to either stl or 3mf format, upon importing the object into another tool it's as if the array tool was not applied at all. 

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.2
[/code]
```

### Subproject(s) affected?

Draft

### Anything else?

Applying a compound or connect operation after the array tool then exporting gives the result I would expect from just exporting the array object. Importing an exported 3mf of an array object into prusa slicer shows the array subelements are placed directly on top of each other, so possibly the array intervals are not being applied on export to the subelements?

Test case of a 3x3 array of cubes, exporting the part called "failure" gives an incorrect result, the two "success" parts work as expected.
[arrayexportbug.zip](https://github.com/FreeCAD/FreeCAD/files/12517045/arrayexportbug.zip)

Result of importing the failure and success parts exported as 3mf into prusa slicer, with the 9 stacked failure subelements selected.
[image]

Result when importing exported stl's for the same parts.
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-09-07:
Confirmed. It has to do with the fact that you have created a link array. An easier way to avoid the issue is to uncheck the "Link array" checkbox when creating the array.

The problem also occurs with an array created from a Link, so it is not a Draft issue.

bgbsww, 2023-11-08:
This also happens in the .ast ( Ascii STL ) format, which is great because it makes it really clear what is happening here:  The multiple elements are in fact being exported, but every element in the array has the exact same position; the array element positioning is not being applied during the export.

bgbsww, 2023-11-08:
Here's the issue.  Two problems:  the last parm of getLinkedObject should be true to allow population of the transformation matrix, and the meshCache doesn't know to recompute and use that matrix on subsequent loops.

Disabling the cache makes it work, but that's a lousy solution, so rework of the cache is in order.

https://github.com/FreeCAD/FreeCAD/blob/ba95a534d5e622721db45b48d20b933e85f01138/src/Mod/Mesh/App/Exporter.cpp#L110-L112

closed 2023-11-11 by wwmayer

Montaire, 2023-11-25:
I have the same issue with the 35176-2023-11-23 development version as well.

In FreeCad it looks like this:
[image]

Exported from 21.1 release:
[image]
As you see from the support only one element is exported

Exported from 22  35176-2023-11-23 release:
[image]

And here is my model, I am not sure if I did anything wrong here:
[battery_holder_v3.6.zip](https://github.com/FreeCAD/FreeCAD/files/13464191/battery_holder_v3.6.zip)

bgbsww, 2023-11-25:
At first glance, I think this is a different issue.  The link array is being followed, but the underlying mirror isn't right.  I'm suspicious that it has to do with the (lack of) depth setting in the getLinkedObject call, but I haven't started looking in earnest yet.  Thanks for the example file!  It's clearly an issue.

PR #11322 (2023-11-11): Fix #10545 #11200 by using position of array members
Fix #10545
PR #11539 (2023-11-29): Don't apply the transformations recursively, just make sure the right…
… one is applied to the top subobject.

Fix #11200 Fix #10545   Without nasty side effects as evidenced by @Montaire example files.