FreeCAD/FreeCAD #10549: [Problem] Techdraw Thread-Design

sebaras, 2023-09-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In Techdraw there is a tool to draw a thread. The problem is that these threads do not comply with the ISO standard in various respects. This applies to external and internal threads. The problem is in the red marked area, as well as with the chamfer in the yellow marked area.
[image]
[image]

### Full version info

```shell
[code]
OS: Windows 10 build 19042
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Switzerland (de_CH)
Installed mods: 
  * A2plus 0.4.60n
  * Assembly4 0.50.2
  * CADExchanger
  * ExplodedAssembly
  * fasteners 0.4.60
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-09-05:
Please elaborate specifically how they don't comply

WandererFan, 2023-09-05:
@edi271, any thoughts?

grubertechnic, 2023-09-05:
Duplicate from #6127

edi271, 2023-09-06:
I think this is no issue of CAD. At this point the DIN standard is exaggerated.
A technical drawing is allways a work instruction how to create the object. In the shown case it says: "take a distinct thread cutter, and cut a thread having she shown depth". How the end of the thread is looking like is not a point of interest. Anyone can just create the shape his thread cutter has.

pitch-circle, 2023-09-08:
> The problem is in the red marked area, as well as with the chamfer in the yellow marked area.

Would it be possible to quote the source of your image please?

I am under the impression that it is ISO6410-3 2021 (simplified representation) that applies here.

A preview of that Standard can be found here:

https://cdn.standards.iteh.ai/samples/81378/45c613c4e4bf433495b54bdea7f1a878/ISO-6410-3-2021.pdf

Figure 2 of the Standard seems to relate to your complaint and it does not quite match what is shown in the extract you have attached.

Sure, TechDraw is not matching the ISO, but is it necessary that TD does so?

Only if we answer yes to that do we need to proceed further. And if we do proceed, we need to establish just what would make it match. I think the chamfer in yellow on which this issue is based is itself not correct.

WandererFan, 2023-09-08:
If I put the TechDraw image up beside the ISO6410 image, I see these differences:
The major diameter line is dashed, but should be solid.
The major diameter line is not joined to the line showing depth of thread.
The point of the hole is squared off, but should be a point.
There is no dash dot line for the centre line of the hole.

[image]

WandererFan, 2023-09-08:
And one more.  The major diameter line should be continuous, not dashed.

WandererFan, 2023-09-17:
#5766 is also about threads and their representation.

luzpaz, 2023-09-17:
Should these tickets be consolidated ?

WandererFan, 2023-09-17:
Yes, I think so.  They are all about threads and their representation in the drawing.

luzpaz, 2023-09-18:
Closing as this ticket has been consolidated into #6127

closed 2023-09-18 by luzpaz