FreeCAD/FreeCAD #10554: [Problem] Legacy DXF importer with Draft-dxf-importer V1.41 breaks KiCadStepUp WB

DeepCoreSystem, 2023-09-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The V1.41 DXF importer library breaks the  [KiCadStepUpMod by easyw](https://github.com/easyw/kicadStepUpMod/issues/181)

Trying to load a PCB silkscreen in DXF format yields the error `cannot unpack non-iterable bool object`

In [my comment](https://github.com/easyw/kicadStepUpMod/issues/181#issuecomment-1706299022) I have a rather ugly workaround for the time beeing.

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly3 0.12.0
  * fasteners 0.4.59
  * InventorLoader 1.4.0
  * kicadStepUpMod 10.19.4
  * Manipulator 1.5.0
  * sheetmetal 0.2.63
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

```
17:48:04  opening /home/pde/Downloads/test/test-F_Silkscreen.dxf...
17:48:04  There has been an error:
17:48:04  Unable to identify DXF file version
17:48:04  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/tmp/.mount_FreeCAXl4oNg/usr/Mod/Draft/importDXF.py", line 2766, in insert
    processdxf(doc, filename)
  File "/tmp/.mount_FreeCAXl4oNg/usr/Mod/Draft/importDXF.py", line 2136, in processdxf
    drawing = dxfReader.readDXF(filename)
  File "/home/pde/.local/share/FreeCAD/Macro/dxfReader.py", line 399, in readDXF
    (infile, drawing) = sm.run((infile, None))
<class 'TypeError'>: cannot unpack non-iterable bool object
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

easyw, 2023-09-06:
@yorikvanhavre 
the Legacy importer is broken with all KiCAD dxf generated files... it is not related only to my kicadStepUp WB, but to load a generic dxf file formed as the one generated by kicad.

luzpaz, 2023-09-07:
CC @kpemartin

DeepCoreSystem, 2023-09-07:
Made a [pull request](https://github.com/yorikvanhavre/Draft-dxf-importer/pull/26) to fix the legacy DXF import

luzpaz, 2023-10-09:
@DeepCoreSystem PR was merged. is there anything left to do here ?

DeepCoreSystem, 2023-10-10:
It works on a new install of FreeCad 0.21.1!
So I should close this issue, shouldn't I?

closed 2023-10-10 by DeepCoreSystem