FreeCAD/FreeCAD #10580: [Problem] The "Start" configuration is no longer realized.

Almisuifre, 2023-09-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Launch FreeCAD, go to Edit>Preferences>Start>Start Page Options>Options
In "Switch to workshop after loading", Set to "Part Design".
Click on "Apply" and then on "Ok".
Close the software.
When you open the software again, the Part Design workshop does not start.

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: French/France (fr_FR)
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

Just to say that it worked on the previous version.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-09-06:
I think you're misinterpreting what you're looking at, the Start refers to the Start Workbench which is triggered by a different setting as per screenshot below:

[image]

You want the `Start up workbench` at the bottom of the page to be set to Part Design which will automatically tick the Autoload checkbox, it has to by default. The setting you were referring to is after the user click Load File or Create New in the Start Workbench then the user can have another workbench load in it's place and close the Start Wb down.

See https://wiki.freecad.org/Start_Preferences

Roy-043, 2023-09-07:
The Start preference "Switch workbench after loading" is no longer used AFAICT. Better to remove it from the GUI then.

Almisuifre, 2023-09-07:
> I think you're misinterpreting what you're looking at, the Start refers to the Start Workbench which is triggered by a different setting as per screenshot below:
>
> [image]
>
> You want the `Start up [...]
Ok, I do not know what you wanted to do. Indeed, it would be good to remove it from the GUI as it was indicated just below you.

It’s actually a duplicate.

On the other hand it is still great to be able to directly access the right workshop.

Roy-043, 2023-09-07:
OK I was confused as well. Now that everything has been cleared up, we can close this one.

closed 2023-09-07 by Roy-043