FreeCAD/FreeCAD #10585: [Problem] Path: Mill Face does not create a path in certain circumstances, nor update paths in all circumstances

VectorHasting, 2023-09-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There appears to be some situations where the Mill Face operation does not produce a path, and others where it will not update its path, making the path appear to be drawn incorrectly. 

Steps to Reproduce these Problems:

In Part Design:
1) Create a square box centered on the XY Plane origin: 60mm on a side.
       a) Pad it to 9.05mm
       b) Mirror it across the XY plane (Note, this is not necessary for creating all the following problems)
2) Switch to Path Workbench.
    (The Tool used was Type endmill, Cutting Edge Height 6.35 mm, Diameter 25.4mm, Shank Diameter 6mm, Length 80.9mm, Chipload .2mm, flutes 3, Material HSS, SpindleDirection: Forward.)
3) Click: Create a Job
 !!! Note: when selecting the object on which to create the Job, there is only one option: Body 0, but when it is clicked, it changes to Body 1. In previous runs trying to reproduce these errors, I repeatedly encountered the error Job.ERROR: 'PySide2.QtWidgets.QListWidgetItem' object has no attribute 'setModel'  With these instructions, however, I did not encounter that error. 
    Click OK
4) Reselect the Job Object: go to the Setup Tab:
    a) It should default to "Extending Model's Bounding Box"
    b) leave all options as is: **specifically**, leave both **Ext. Z at 1mm**
    c) Select a corner of the Stock Object and click "Set Origin"
    d) Click OK.
5) For clarity, hide the body in the tree, and show the Model-Body and the Stock.
6) Add the spoiler board tool (described above) to the job's tool controllers
7) Select the top of the Model-Body in the 3D display, and click the Mill Face Button -> select the spoilerboard tool.
    a) On the Mill-Face operation Tab:
       i) choose Boundary Shape: **Perimeter** (this is key to creating this bug), Alternatively, choose Boundbox
       ii) choose Cut Mode Conventional (this does not seem to affect this bug),
       iii) Leave Pattern at ZigZag
       iv) Leave Angle 45 deg.
       v) Choose Step Over Percent 50
       vi) Leave Material Allowance at 0
       vii) choose Clear Edges (this does not seem to affect this bug)
    b) Click OK
    (you should be back in the tree)
8) In the tree, open Operations, select Millface and modify these properties:
    a) Depth > StepDown: This value is inappropriate for spoilerboard tools. For this example, set it to 1 

Problem 1 should now be reproduced: 

Observed behavior: There is no path whatsoever. 
Expected behavior: It should generate a milling path. 

Users can work around this bug by deselecting Boundary Shape "Perimeter" or "Boundbox" or "Face Region" and using: "Stock"

Note: If the object is not mirrored, then both "Face Region" and "Stock" produce paths. However, it is unclear to this user why each of those create the different paths that they do. One hypothesis (without the benefit of reading the code) is that Face may be using the tool's center, and drawing the face region, and Stock might be using the tool's diameter, and drawing the Stock region.  This might not have been readily apparent during testing without using such a large diameter tool. 

To reproduce Problem 2, follow these additional steps:

9) Click the Stock Object, and change the Ext Zneg and Ext Zpos to 9.5mm

Problem 2 is now reproduced:

Observed behavior: The path does not update for the deeper path it needs to create.
Expected behavior: As in other contexts, it should do an active recalculation, or show markings in the tree that a recalculation is pending. 

Work around: One must select MillFace and click "recompute object." Sometimes this is insufficient, and it is necessary to select the Job and mark it for recompute, then perform the recompute. 

Thank you!

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.12.3
  * fcgear 1.0.0
  * ThreadProfile 1.82.0
[/code]
```

### Subproject(s) affected?

None

### Anything else?

This problem is discussed here on the forum: [https://forum.freecad.org/viewtopic.php?style=3&t=76912](url)

I have had an additional problems with the path not being created down to the face of the object, but always being 1mm short of it. This was a much more complex shape, but I was not able to reproduce that behavior. Moreover, restarting FreeCAD seems to cause it to disappear. So, it's a ghosty kinda bug. That is discussed in this thread: [https://forum.freecad.org/viewtopic.php?p=704746#p704746](url)

[MillFaceIssue_02.zip](https://github.com/FreeCAD/FreeCAD/files/12545202/MillFaceIssue_02.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-09-07:
@VectorHasting thank you for the very thorough ticket. Some feedback: you're not using the markdown hyperlink formatting correctly. Scroll back to your ticket and hover over the forum links to see. Checkout https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#links

VectorHasting, 2023-09-07:
@luzpaz, Sorry, and thank you. I think I see what I did wrong and I believe I got it now. 

Will try it here: 

I see you gave this to @Sliptonic... the other issue I opened: #10584 references [his how-to-video from version .17](https://www.youtube.com/watch?v=k1ezXIaOw60) .  If you haven't already, he might be a good person to at least give a look at that issue as well. 

Thanks again.

life5ign, 2023-11-24:
I seem to be having this issue too; just watched all of @sliptonic 's excellent 0.17 videos (I'm using 0.21.0 now), and MillFace is not creating a path.  I am successfully creating a Profile path, which I can simulate successfully too.

life5ign, 2023-11-24:
Just found out that changing "Boundary Shape" in the "Operation" section of the MillFace Operation settings to either Face Region or Stock successfully created the path.  The default is Bounding Box, which seems like should produce the same path, but I don't fully understand all the different levels/heights/boundaries yet.  Anyway, the default should ideally work though, for a simple shape like a cube (I'm using a PartDesign Sketch rectangle padded, with a smaller rectangle Skech pocketed out of the center).

JohnOCFII, 2025-01-21:
Seems like this is still an issue based on the last comment and it's linked issue.

tarman3, 2025-08-02:
Can not reproduce
Please attach FCStd file with simple example
Also write short instruction

sliptonic, 2025-10-04:
I can't confirm.   MillFace has some fundamental problems as noted in issue #24290.  
AFAIK, there's no specific bug right now that would cause this problem but it is very possible that the specific symptoms reported could be triggered by either an unknown bug or user configuration problems.  (e.g. using a tool that is too large for the anticipated clearing area.

The re-implemented facing operation is designed to address all of these things with a cleaner implementation.  It will not be merged until after the 1.1 release.

As far as I'm concerned, this issue can be closed and reopened if OP can provide a clean test case.  It would be nice to have that for the re-implmented version too.

closed 2025-11-28 by chennes