FreeCAD/FreeCAD #10588: [Problem] SystemDllWrite violation reported by Cylance Protect

gilmiz, 2023-09-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Cylance Protect identify **SystemDllWrite** as threat and block it.
No apparent influence on the way Freecad operate other than these notifications.

[image]

[image]

Description as for why can be found in the [Cylance API / User API Guide](https://docs.blackberry.com/content/dam/docs-blackberry-com/release-pdfs/en/cylance-products/api-and-developer-guides/Cylance_User_API_Guide.pdf) (page 76), described as: 

> systemdllwrite (System DLL Overwrite) - Windows only: An attempt to overwrite a system DLL has been detected.

The issue was not notice on the 0.20 version.

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Danish/Denmark (da_DK)
Installed mods: 
  * Assembly4 0.50.2
  * fasteners 0.4.64
  * sheetmetal 0.3.0
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-09-07:
Needs to be flagged as false positive

wwmayer, 2023-09-10:
The function `SystemDllWrite` isn't used anywhere in the FreeCAD code base.

luzpaz, 2023-09-11:
@gilmiz where did you download FreeCAD from ?

gilmiz, 2023-09-11:
It was downloaded form the freecadweb.org site

luzpaz, 2023-09-14:
Weird, is it coming from one of the addons ?? 
CC @shaise and @Zolko-123 have you seen this before ? (`SystemDllWrite`)

Zolko-123, 2023-09-14:
Assembly4 is Python only, no DLL involved

wwmayer, 2023-09-15:
Each FreeCAD add-on is Python only, so no DLL involved. The only theoretical possibility is that a 3rd party library uses this function. But I don't think this is the case because if you google for "SystemDllWrite" then there are altogether only three hits and one of them is this ticket.

luzpaz, 2023-09-15:
@gilmiz what is the exact download link that you used ? 

(Can the issue be coming from one of the Window installers, for example the libpack related windows installer ?)

shaise, 2023-09-15:
> Weird, is it coming from one of the addons ?? 
> >  * Assembly4 0.50.2
> >  * fasteners 0.4.64
> >  * sheetmetal 0.3.0
> 
> CC @shaise and @Zolko-123 have you seen this before ? (`SystemDllWrite`)
> 

Did not encounter this issue. But then again I do not use Cylance Protect.

luzpaz, 2023-10-09:
@gilmiz any follow-up to this ? we haven't heard of any other reports of this btw.

gilmiz, 2023-10-13:
Apologies. 
It was downloaded from the www.freecadweb.org (https://www.freecad.org/thankyou.php?url=https%3A%2F%2Fgithub.com%2FFreeCAD%2FFreeCAD%2Freleases%2Fdownload%2F0.21.1%2FFreeCAD-0.21.1-WIN-x64-installer-1.exe)

luzpaz, 2023-10-15:
Can you send Cylance feedback on this, @gilmiz ? This looks like a false positive. We haven't heard of any other issues since.

I'm closing ticket for now. Please feel free to re-open it.

closed 2023-10-15 by luzpaz