FreeCAD/FreeCAD #10589: FreeCAD switching File when confirming/pressing tab/double clicking

EMPRSR, 2023-09-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This problem only occurs when multiple files are in the 'Combo View' ('Tree View') and happens often when a value in the 'Property Editor' is changed and confirmed by pressing Enter or Tab. It automatically switches to the first opened document unrelated to the arrangement in the tab row in the 'Main View Area'.

The issue also happens every time when for example a sketch or 'Part Design' operation in the 'Combo View' is double clicked for editing.

A full reinstall of FreeCAD did not help despite using the brew uninstaller and manually hunting any remaining files.
Furthermore, deactivating the Treeview Actions (such as Sync View) has no effect.

Using FreeCAD Version 0.21.0 (33675) on macOS 13.

### Full version info

```shell
[code]
OS: macOS 13.0
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * freecad.gears 1.0.0
  * FeedsAndSpeeds 0.5.0
  * Dracula 0.0.5
  * Manipulator 1.5.0
  * fasteners 0.4.63
  * CurvedShapes 1.0.4
  * Help 1.0.3
  * Curves 0.6.13
  * Assembly3 0.12.0
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

henrebotha, 2024-01-19:
I am on macOS 12.7.2 and FreeCAD 0.21.1, and I'm getting the same behaviour. It happens in FreeCAD Link too ([this version](https://github.com/realthunder/FreeCAD/commit/096210d21183e9dfdc3b25777760bfb6c00a210b)).

henrebotha, 2024-01-31:
FWIW this is a massive UX problem and almost makes the software unusable.

3rduino, 2024-02-10:
Same for me on MacOS 14.3 (Sonoma) and FreeCAD 21.2

Achu69, 2024-03-10:
Same issue, using. The only possible work around is not to use enter/tab when editing data parameters and just click out of the box instead or only have one model open i.e. close all other tabs, including Start Page.

OS: macOS 14.4
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36331 (Git)
Build type: Release
Branch: main
Hash: eb5a1b333946ce4ac8a4ead835cba8bdcf602a87
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)

luzpaz, 2024-08-22:
@JohnOCFII do you mind trying to reproduce ?

Achu69, 2024-08-22:
Hi, sure.

My current system:
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38459 (Git)
Build type: Release
Branch: main
Hash: 58729b3ab0060a1525fc45c87beabfd4b3d5e22e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods:
  * Gridfinity 0.5.0
  * A2plus 0.4.65
  * fasteners 0.5.26
  * LCInterlocking
  * Curves 0.6.42

The issue:

Open two files:

***@***.***

Select the one on the right for working

***@***.***

Create some geometry and pad it. In this case a circle with 10mm pad and click ok. Select the pad operation in the tree

***@***.***

Try to change the pad thickness via data tab • Length to say 15mm and press <Enter>

***@***.***

***@***.***

The working model gets switched to the one on left

***@***.***

This also happen if instead of pressing <Enter> you press <Tab>. The working file changes only from right to left until it gets to the first tab on the left.

Hope this helps and please let me if I can help further.

Regards
Ed.

luzpaz, 2024-08-22:
@Achu69 unfortunately responding via email b0rks the formatting and excludes attachments. It's better to respond directly on github

Achu69, 2024-08-22:
Sorry, here it is again, and thank you for your dedication to this great tool!
 
My current system:
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38459 (Git)
Build type: Release
Branch: main
Hash: 58729b3ab0060a1525fc45c87beabfd4b3d5e22e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods:
  * Gridfinity 0.5.0
  * A2plus 0.4.65
  * fasteners 0.5.26
  * LCInterlocking
  * Curves 0.6.42
 
The issue:
 
Open two files:

[image]

Select the one on the right for working

[image]

Create some geometry and pad it. In this case, draw a circle with 10mm pad and click ok. Select the pad operation in the tree

[image]

Try to change the pad thickness via data tab --> Length to say 15mm and press Enter

[image]
[image]

The working model gets switched to the one on left

[image]

This will also happen if, instead of pressing Enter, you press Tab. The working file changes only from right to left until it gets to the first tab on the left.

JohnOCFII, 2024-08-22:
> @JohnOCFII do you mind trying to reproduce ?

I can reproduce this problem with the most recent functioning weekly build (38459), but the problem is resolved with the change added in PR# 15944.

closed 2024-08-25 by chennes

AustinPackard, 2026-10-27:
Similar issues are happening on this RC build:

OS: macOS 27.2
Architecture: arm64
Version: 26.3.0RC2.20261018 (Git shallow)
Build date: 2026/10/18 05:49:18
Build type: Release
Branch: (HEAD detached at 9fb16e7f9)
Hash: 9fb16e7f981adb09f7d7f130ce085d68e9825358
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenTheme.qss/OpenDark/
QPA/File dialog/Color dialog: cocoa/via Qt/via Qt
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball Classic/Object center
Logical DPI/Physical DPI/Pixel Ratio: 72/112/2
OpenGL version/vendor/renderer: 2.1 Metal - 91.7/Apple/Apple M5
Installed mods:
  * History 0.2.1
  * OpenTheme 2026.10.5

Specifically, I can most reliable reproduce this issue when toggling visibility of a body or the origin of a body.