FreeCAD/FreeCAD #10598: [Problem] Sketcher_Dimension can't constrain angle for arc

qewer33, 2023-09-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently when you click on an arc with the Sketcher_Dimension tool active, you can only cycle between the radius and diameter constraints. The angle constraint also works on arcs so this should be added as well. Also when an arc's radius or diameter is already constrained, the Sketcher_Dimension tool should default to an angle constrain.

### Full version info

```shell
OS: Fedora Linux 38 (KDE Plasma) (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34177 (Git)
Build type: Release
Branch: master
Hash: 331b76485002084ab341cfe5b27eb492b50a9c85
Python 3.11.4, Qt 5.15.10, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4.backup1691951820.225657 0.50.2
  * Assembly4
  * Behave-Dark-Colors 0.1.1
  * Curves 0.6.13
  * DynamicData 2.46.0
  * fasteners 0.4.59
  * Freecad-Built-in-themes-beta 1.2.2
  * freecad.gears.backup1692223246.5590403 1.0.0
  * freecad.gears
  * ExtremeProLight 1.6.0
  * A2plus 0.4.60n
  * ExplodedAssembly
  * Glass
  * kicadStepUpMod 10.19.4
  * Silk 0.1.4
  * lattice2 1.0.0
  * Manipulator 1.5.0
  * Leaf Dark Colors 0.1.1
  * FreeCAD_DarkPreferencePack 0.0.1
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2023-09-08:
Good idea!

maxwxyz, 2024-01-04:
Should be possible if there is a fix for #11164

closed 2024-01-07 by qewer33