FreeCAD/FreeCAD #10601: [Problem] No visibility or configuration available for the formula editor hotkey

atlast-50, 2023-09-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On textboxes that allow the use of expressions there is a button that you can click to open the formula editor. The tooltips for this button only state "Enter an expression..." and give no indication that "=" is a hotkey to open up the formula editor. Additionally, this hotkey is neither listed nor configurable in the Tools > Customize menu

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.2
[/code]
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-09-24:
@atlast-50 what do you propose ? (mockups would be an extra bonus)

atlast-50, 2023-09-24:
@luzpaz My thought was just to mimic the other tooltips that show hotkeys. So in sketcher for instance constrain coincident's tooltip is (assuming default keybindings)
**Constrain coincident** (C)
Create a coincident constraint between...

So just change the tooltip for the formula editor button from "Enter an expression..."/"Expression: EXPRESSION" to something like

**Formula Editor** (=)
"Expression: EXPRESSION" or "Enter an expression" if none is set. 

Where EXPRESSION is whatever the expression is set to be. And then also give the ability to change this hotkey.

atlast-50, 2023-09-24:
Possibly also it might be worth considering renaming the window from "Formula Editor" to "Expression Editor"? I'm new to freecad, and cad software in general, but when I was trying to find the hotkey for the "Enter an expression..." button my path was check tooltip -> search through hotkey configuration -> google looking for an "Expression hotkey" because I didn't read the title of the window that pops up to notice that it's called "Formula Editor", and this gave no helpful results -> more random googling until I found it.

Everywhere else that I can see just refers to expressions as expressions, if this window is the only one that refers to them as formulas it would make sense to rename that to be consistent.

yorikvanhavre, 2023-09-25:
I agree "formula" should be changed to "expression" everywhere in FreeCAD. Expression is our official term here and it's the same mechanism used throughout FreeCAD.
 Indicating the (=) shortcut in the tooltip is a good idea... I would not make it customizable for now because it's a default in many apps (spreadsheets etc) and unless someone asks for it  I don't think it's worth the effort.
This is a good first issue for potential new contributors... @atlast-50 would you be interested? :innocent: We can help ;)

commit c8fbe03e (2023-10-16): Gui: Rename forumula to expression + add hotkey to expression tooltip

Fixes #10601
closed 2023-10-16 by yorikvanhavre