FreeCAD/FreeCAD #10604: [Bug] Sketcher: If a DSH is active and user use constraint shortcuts

PaddleStroke, 2023-09-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

 If :
- a DSH is active 
- Some geometry are selected (this happens only with Sketcher_Dimension DSH right now)
- User use constraint shortcuts
- > The constraint is create while the DSH is still active.

To test : 
- draw a line.
- Start sketcher_dimension and select the line
- Press K, D
- > a distance constraint is created while Sketcher_dimension is still active.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-01-21:
This still happens

luzpaz, 2025-06-08:
Who wants to tackle this ?

PaddleStroke, 2025-12-16:
still relevant.

closed 2026-09-11 by maxwxyz