FreeCAD/FreeCAD #10605: [Problem] Sketcher_Dimension does not offer to apply angle to arcs.

PaddleStroke, 2023-09-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Sketcher_Dimension should offer to apply angle to arcs as a secondary mode (after radius/diameter).

Ideally if the arc already has a radius/diameter, then it should offer only angle. But this is a broader issue for all constraints. If a line already has a distance, then it would be best no to offer distance.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2023-09-08:
@PaddleStroke What about arc length? Is that doable? It has been requested a few times in the past.

PaddleStroke, 2023-09-08:
That's a separate issue because right now we don't have arclength as a constraint. So before we can integrate it in sketcher_dimension, it needs to be implemented as a constraint. Which is a similar job than what @FlachyJoe made with circle-circle distances and others.

closed 2023-09-19 by chennes

maxwxyz, 2023-10-22:
@PaddleStroke is it possible to offer the arc angle dimension when selecting the two points defining the arc or three points when selecting the arc center? Currently I have to cycle with "M" through the offers of the tool (radius, diameter, angle).

Would it be possible if the arc already has a radius or diameter that in this case with the first click the angle dimension is activated?

PaddleStroke, 2023-10-22:
All your suggestions are possible.

We need to choose one then. Do you perhaps know how other CAD softwares handle this case ? I think it is best to be consistent with other softwares than reinventing our own standard. Unless its better.

maxwxyz, 2023-10-23:
I would prefer the following: If the radius of the arc is defined (first click of the tool on the arc line), another click on the arc line should activate the angle tool. Then I stillI have the possibility to define the length between the points / horizontal or vertical distance by clicking on both points of the arc.
If it would be possible, the angle could also be presented when clicking on both arc points and the center point. You could in general invoke the angle tool when selecting three points, as an angle can be defined by three points.

PaddleStroke, 2023-10-24:
Ok. Please open an issue about this so that it doesn't get lost. I'll look into it asap. It shouldn't be too hard but I wonder how to handle the case where the arc radius is constrained but not directly by a radius or a diameter.