FreeCAD/FreeCAD #10607: [Problem]  Preselection and Selection Colors are not separated between 3d View  and sketch editing

obelisk79, 2023-09-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD currently applies the same preselet and select colors for both geometry and sketch lines. The problem is that colors which are accessible and work well in 3d view generally do not meet ACPA color standards for accessibility while in sketch view. 
[image]

in 3d view the above colors look appropriate
[image]

in sketch view they do not
[image]

### Full version info

```shell
[code]
OS: EndeavourOS (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * ExtremeProDark 2.9.0
  * persistenttoolbars
  * Defeaturing 1.2.0
  * ProDarkThemePreferencePack 1.0.0
  * Manipulator 1.5.0
  * A2plus 0.4.60n
  * TabBar (Disabled)
  * IconThemes
  * FreeCAD_DarkPreferencePack 0.0.1
  * FreeCAD_DarkPreferencePack_v0.41.zip
  * CubeMenu
  * Dracula 0.0.5
  * Render 2023.8.18
  * FreeCAD_DarkPreferencePack_v0.4.zip
  * SelectorToolbar (Disabled)
  * Behave-Dark-Colors 0.1.1
  * 3DfindIT 1.2.0
  * Curves.backup1692576339.4110856 0.6.13 (Disabled)
  * lattice2 1.0.0
  * Curves 0.6.13
  * Glass (Disabled)
  * freecad.gears 1.0.0
  * fasteners 0.4.63
[/code]
```

### Subproject(s) affected?

None

### Anything else?

My proposal would be to create new color preferences in Sketcher WB which separate pre-select, and selection color for sketch elements while in edit sketch mode.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

obelisk79, 2023-09-08:
@PaddleStroke @abdullahtahiriyo 
I'd like your feedback on whether or not my proposal is possible and reasonable.

luzpaz, 2023-09-23:
bumping for visibility

PaddleStroke, 2023-10-26:
The colors look good in 3d view. But in sketcher the selected color is very bad for me (red colorblind).
I am not sure how easy/hard customizing the selection colors is.

@wwmayer what do you think? Is it feasable to add preferences for selection when selecting in edit-mode?

obelisk79, 2023-10-26:
Can this not be something handled directly in sketcher edit mode?

abdullahtahiriyo, 2023-10-27:
You can change the colors used for selection and preselection at the sketcher via parameters:
"HighlightColor"
"SelectionColor"

What it is not currently possible is to set the PreselectSelectedColor. It is straightforward to add it to ```EditModeCoinManager::ParameterObserver```

The part I did not quite understand is that in the screenshot above, in View there are this selection/preselection. But in my FC there is no such thing. Is this an example of what is expected as feature?

How much does it help this?

abdullahtahiriyo, 2023-10-27:
[image]

abdullahtahiriyo, 2023-10-27:
It would be great to have some kind of profiles, and one for each of the different types of colorblind. I think it would be good to have something "FreeCAD wide". For sketcher-wide, if I understood it correctly, it is as easy as adding the preferences to settings.

closed 2025-02-14 by github-actions[bot]