FreeCAD/FreeCAD #10612: [Problem] Rotate view should stop rotating if I release the mouse button

pvginkel, 2023-09-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps are:

- Hold down Shift.
- Right click down.
- Move the mouse. This rotates the view.
- Release the mouse.
- The view no still rotates.
- Release shift. The view stops rotating.

I think it'd be more user friendly if rotation stops as soon as you release the mouse.

### Full version info

```shell
[code]
OS: Windows 11 build 23536
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods: 
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

qewer33, 2023-09-08:
That's how the CAD navigation style works. Try a different one if you don't like it's behavior (there's a navigation style selector in the bottom right of the application window):
[image]

obelisk79, 2023-09-08:
I agree with Qewer. The behavior matches the navigation style.

pvginkel, 2023-09-09:
Apologies. I checked the documentation and code, and you're absolutely right. I checked a few of the other navigation styles and I find them all quite unintuitive. I appreciate thought that that is because I have very little experience with CAD products (but I do have programming experience). I raised this because I see this kind of subtle UX interaction go wrong often and when you get used to it, you lose sight of that it may be strange. Newcomers are more aware. The UX I described is basically how drag and copy files in Windows works. Windows only looks at the keyboard state the moment you press down the mouse. It modifies the interpretation of the mouse pressed event only. So, I was expecting that behavior and had a lot of difficulty getting used to FreeCADs behavior. I'll try a few of the navigation styles. See if one makes sense to me.

closed 2023-09-09 by pvginkel

obelisk79, 2023-09-09:
No worries, I find the tinker cad navigation style to feel pretty natural to use. Maybe give that one a test drive in your experimentation. There's no shortage of options though.

pvginkel, 2023-09-10:
I did. I'm finding this one quite intuitive. Thank you for the suggestion.