FreeCAD/FreeCAD #10614: [Problem] Selected and hovered faces in the 3D view shouldn't be shaded

qewer33, 2023-09-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently the 3D view selection colors looks way off from the colors defined in `Preferences > General > Selection`. The colors also change depending on the angle of the viewport. This is because the selected and hovered faces are shaded just like the rest of the part. Below is an example that shows the issue:

[image]
[image]

In both images the same face is hovered but the bottom one looks barely highlighted because of the shading and the view angle. Other CAD software solve this issue by displaying selected and hovered faces as non-shaded (or applying "less shading" to those faces). Below is an example from OnShape:

[image]
[image]

I've done a proof of concept implementation of this in FreeCAD and can open a PR if people are in favor of this. It might also make sense to implement this as a preference but I'm not sure how to approach that. Open to discussion.

[image]

### Full version info

```shell
OS: Fedora Linux 38 (KDE Plasma) (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34177 (Git)
Build type: Release
Branch: master
Hash: 331b76485002084ab341cfe5b27eb492b50a9c85
Python 3.11.4, Qt 5.15.10, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4.backup1691951820.225657 0.50.2
  * Assembly4
  * Behave-Dark-Colors 0.1.1
  * Curves 0.6.13
  * DynamicData 2.46.0
  * fasteners 0.4.59
  * Freecad-Built-in-themes-beta 1.2.2
  * freecad.gears.backup1692223246.5590403 1.0.0
  * freecad.gears
  * ExtremeProLight 1.6.0
  * A2plus 0.4.60n
  * ExplodedAssembly
  * Glass
  * kicadStepUpMod 10.19.4
  * Silk 0.1.4
  * lattice2 1.0.0
  * Manipulator 1.5.0
  * Leaf Dark Colors 0.1.1
  * FreeCAD_DarkPreferencePack 0.0.1
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-09-09:
@qewer33 Can you please report back a link to your branch so I can test before giving some feedback, it's not something that's bothered me and I don't remember it being raised by either users new or dyed in the wool before.

qewer33, 2023-09-09:
> @qewer33 Can you please report back a link to your branch so I can test before giving some feedback

Just pushed to: https://github.com/qewer33/FreeCAD/tree/selection-no-shading

> it's not something that's bothered me and I don't remember it being raised by either users new or dyed in the wool before.

Well it's an issue that has bothered me for a long time since I started using FreeCAD. And even worse, with certain material properties, the selection and highlight colors are just straight up invisible.

Syres916, 2023-09-09:
Thanks for the link, yep, I've no qualms with that. I'm doing some CNC testing related to another issue and tried different drawing styles all of which I could still easily select and see what I needed to and cannot see any detrimental effects so it's fine with me.

kadet1090, 2023-09-09:
It'd be best to combine both styles, like shaded + ~80% of plain color on top, but this possibly could be hard to implement. I think however that disabling shading would be step in right direction.

obelisk79, 2023-09-09:
It was mentioned on discord that this would be great if the color could just be overlayed on the model with transparency so you still get shading but it isn't washed out by the omni light in the 3d view. I would very much like to see that. the problem I find with this solution is that it washes out too much visual detail in exchange for clear coloring.

qewer33, 2023-09-18:
I'd like to continue working on this but my OpenGL skills aren't good enough to implement the "slightly shaded, mostly base color" solution proposed in the comments above. I also agree, that would be a much better solution so I don't want to open a PR with my current implementation (which just disables shading).

@wwmayer (I don't know anyone else who's familiar with the Gui/Rendering code) would be awesome if you looked at this when you have time. Does it seem feasible?

wwmayer, 2023-09-19:
I don't know if it's possible to only do a bit of shading. What you can try is to test the methods of SoLazyElement like setting ambient, emissive, specular or diffuse color or changing the shininess. These are the options you have to customize the rendering.

luzpaz, 2023-11-06:
@qewer33 have you made any progress on this ?

qewer33, 2023-11-12:
@luzpaz nope, my old branch still exists but it isn't ready for a PR

This issue can probably be closed as not planned unless someone else wants to work on it

luzpaz, 2023-11-28:
@Syres916 any interest taking over this attempt to improve the UI/UX ?

Syres916, 2023-11-29:
I think/near certain this is well outside my skills but thanks for asking.

pierreporte, 2024-04-20:
Is this fixable since the introduction of new appearance parameters with materials revamp?

maxwxyz, 2024-04-21:
@davesrocketshop ^FYI

davesrocketshop, 2024-04-21:
> @davesrocketshop ^FYI

Assign it to me but it's going to depend on other priorities to make it in to 1.0. In the meantime @qewer33 please don't delete that branch

maxwxyz, 2024-07-04:
I'm in favor of these suggestions but there should be a fix for https://github.com/FreeCAD/FreeCAD/issues/13340 as well as if you select the body to apply a material, every face is selected and with a fix to this issue here you would just see the selection color, not the material preview.

maxwxyz, 2025-02-27:
IMO it is not also an issue with shading, but that the color of the selection or preselection is overlayed on top of the underlying face color. So the selection color appears completely different, depending on the objects color - which is bad.

https://github.com/user-attachments/assets/97d9d87c-d546-465e-b652-fab15006bb5c

pierreporte, 2025-02-27:
It's a different problem IMO. There is a shading problem regardless even though it was reduced by the new 3-point lighting. The overlay is a completely different issue.

maxwxyz, 2025-05-21:
indeen, not shading is the issue, but the overlay mode of the color. @kadet1090 is this adressed maybe in relation to transparent previews?

maxwxyz, 2026-08-09:
The current situation is much better, should this issue be closed?

luzpaz, 2026-09-20:
@tritao does you FEP regarding selection touch this issue ?

tritao, 2026-09-20:
> [@tritao](https://github.com/tritao) does you FEP regarding selection touch this issue ?

Which FEP do you mean?

luzpaz, 2026-10-04:
@tritao apologies, was referring to 
* https://github.com/FreeCAD/FPA-grant-proposals/issues/35

R2V1W3, 2026-11-16:
This may not be relevant to this but it is for the issue #34339 marked as duplicate of this, but it applies for whole body selection highlight rather than faces. 

I made an override of selection glare blacking out additional emissive color applied in  **SoBrepFaceSet.cpp** to the whole body/feature selection when selected from the Tree. 

This fixes only B-rep bodies. Eliminating glare in meshes is still work in progress.  

```
bool SoBrepFaceSet::overrideMaterialBinding(
.
.
.
.
    state->push();
    packedColors.clear();
    
// This should eleiminate selection glare
    if (hasPrimary) {
        static const SbColor black(0.0f, 0.0f, 0.0f);
        SoLazyElement::setEmissive(state, &black);
        SoOverrideElement::setEmissiveColorOverride(state, this, true);
        SoOverrideElement::setDiffuseColorOverride(state, this, true);
    }
```
Here are some examples of before and after

[image]

[image]

https://github.com/user-attachments/assets/03a47538-c0e6-4973-aeaa-02c602139f3b

https://github.com/user-attachments/assets/8a969074-2372-42d9-a8ea-6ada5c688571