FreeCAD/FreeCAD #10616: [Problem] Path: Sanity Report is not converting from asciidoc to html

VectorHasting, 2023-09-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

**Error/problem:** 

On Windows systems, the Path_Sanity option (used as a button, at least) does not display the Job Report, instead it displays:

`[WinError 2] The system cannot find the file specified`

**Background**:

I have installed asciidoctor as per [instructions found on their website](https://docs.asciidoctor.org/asciidoctor/latest/install/)

**To reproduce the problem**:

1. Create a Job in Path
2. Specify an output file under Job > Output > Output File
3. Attempt to use the Sanity Check Button. 

Expected Behavior:
Which is [documented on the FreeCAD wiki, at the bottom](https://wiki.freecad.org/Path_Sanity ) 
and [demonstrated in @sliptonic video here](https://www.youtube.com/watch?v=RFu-DQ04uZM&t=903s )

It should:

- generate a Job Report in .asciidoc format at the location specified in Output file, 
- convert that file to .html format, 
- display it in the system's default browser. 

Observed Behavior:

It seems to actually:

- generates a Job Report in .asciidoc format, 
- fail to convert that file to .html format, 
- display the error shown above in FreeCAD's Report View, 
- and launch (on Windows) Microsoft Edge browser, which either displays the home page (if the output file is on a local drive) or an error message (if the file is on a unmapped drive/url server, ie \\fileserver\etc)

**Workaround**:
After installing asciidoctor, manually convert the file using [their instructions here.](https://docs.asciidoctor.org/asciidoctor/latest/get-started/)

Note: the exact command I use is simply:

`asciidoctor  c:\filepath\filename.asciidoc`

It works whether the command is issued from a different directory or different drive. 

Error page screen follows:

[image]

Thank you!

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.12.3
  * fcgear 1.0.0
  * ThreadProfile 1.82.0
[/code]
```

### Subproject(s) affected?

None

### Anything else?

Discussed on the following forum pages:

[Need for manual conversion](https://forum.freecad.org/viewtopic.php?p=594936&hilit=asciidoc#p594936)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-09-09:
If FreeCAD has all the portent data, why can't the Python code just generate the html output itself and completely eliminate the need for yet another dependency? The Start Wb and the AddonManager use generated html and it could even be Stylesheet aware rather than blinding a dark themed user as it appears to do now.
@sliptonic Is there really something special about this function that has to be done through asciidoc?

Syres916, 2023-09-09:
@VectorHasting If I could produce the attached html example straight from the Python Code would you class it as acceptable (I had to zip it to attach here)?

[TwoHex.zip](https://github.com/FreeCAD/FreeCAD/files/12565566/TwoHex.zip)

Syres916, 2023-09-09:
Well I'm about half way through the conversion to be able to generate the html output without the need for a dependency. I'm checking that the fields are dynamic using French as the language so I can see that it's picking up the translations. First commit to branch https://github.com/FreeCAD/FreeCAD/compare/master...Syres916:FreeCAD:Sanity_No_dependencies

[image]

VectorHasting, 2023-09-09:
@Syres916 Thank you for your attention to this! The output looks great :-)  

The zipped file you posted didn't have the picture, so it looks like you've been making fixes between 10 and 5 hours ago ;-) 

This will be nice. Thanks again!

Syres916, 2023-09-10:
@VectorHasting , I have a version for you to test, I'm sure it won't take you long to break it, the images are particularly annoying if the camera has to reset each time, I've added a delay to try to compensate. So copy and paste https://raw.githubusercontent.com/Syres916/FreeCAD/7ee0a5bca0d6e7b106b4dcc56c9447d5ac34d7e9/src/Mod/Path/Path/Main/Gui/Sanity.py into your current builds copy. I've left the code to generate the asciidoc file so if needed yo can send me both the html and the asciidoc so I can compare what's going wrong.

TODO: The Squak icons for Note, Tip, Caution and Warning are to be created and added to the code, that's definitely for later as my artisitc skills aren't great. Maybe we've already got some useful ones buried away in the source already but it will not be in the first PR.
Also the section titles and the Squak responses appear to not be translated which seems a bit strange as they are static text 'pulled' in where required so I'll try and get that done tomorrow, need a break, good luck!!

sliptonic, 2023-09-10:
>  @sliptonic Is there really something special about this function that has to be done through asciidoc?

Nope.  It was done with asciidoc because it was easy.  I would love to see the dependency removed.  It was also my ultimate goal that sanity reports could be generated without the GUI running.  That isn't entirely possible now because of the need to generate the images but please keep the goal in mind while you're reworking this.  

Nice job!

Syres916, 2023-09-11:
I keep hitting a road block with the external stylesheet idea so I'm going to make it all responsive to the style info a the top of the file and possibly give the user some colour options to the preferences but I'm certainly not going overboard. That will be in a follow up PR along with more investigation with regard to getting a reasonable output form the terminal.

closed 2023-09-14 by sliptonic