FreeCAD/FreeCAD #10620: Path Profile `Expand Profile` and `Expand Profile Stepover` missing

EMPRSR, 2023-09-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Creating a 'Profile' operation using the 'Contour'  mode in the Path WB works (despite throwing an error: Please select features from a single solid), however the `Expand Profile` or `Expand Profile Stepover` properties in the Property Editor or Dialogue are missing completely. Activating 'show all' by right clicking on the Property Editor does display new parameters, still the aforementioned expansion settings are not visible. 

Apparently these parameters have already been merged with PR [#3545](https://github.com/FreeCAD/FreeCAD/pull/3545).

The [wiki entry](https://wiki.freecad.org/Path_Profile) for the Path Profile operation shows them in the 'Properties' tab in section 'Profile'.

### Full version info

```shell
[code]
OS: macOS 13.0
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * freecad.gears 1.0.0
  * FeedsAndSpeeds 0.5.0
  * Dracula 0.0.5
  * Manipulator 1.5.0
  * fasteners 0.4.63
  * CurvedShapes 1.0.4
  * Help 1.0.3
  * Curves 0.6.13
  * Assembly3 0.12.0
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

EMPRSR, 2023-09-10:
After doing some more digging around and finally signing up to the forum I was guided to this [thread](https://forum.freecad.org/viewtopic.php?t=78804) and understand that the properties on the wiki are simply not up-to-date.

As development seems to be ongoing I am unsure whether to close this issue or leave it open (as a feature request of sorts).

Furthermore, I want to apologise for prematurely opening this issue and not looking into the forum enough.

luzpaz, 2023-09-11:
CC @sliptonic are documentation efforts for Path ongoing in the wiki or elsewhere ?

luzpaz, 2024-08-23:
> After doing some more digging around and finally signing up to the forum I was guided to this [thread](https://forum.freecad.org/viewtopic.php?t=78804) and understand that the properties on the wiki are simply not up-to-date.

CC @Russ4262 (OP references your thread in the forum)

Russ4262, 2024-08-25:
My opinion is that the wiki needs to be synchronized with the Profile operation's capabilities. The two properties mentioned in this issue are directly related to long-removed capabilities of the current Profile op.

Respectfully,
Russell

mnesarco, 2026-03-22:
@maxwxyz