FreeCAD/FreeCAD #10621: Path Adaptive incompatible with LeadInOut Dressup

EMPRSR, 2023-09-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The 2D Adaptive strategy in the Path WB is fantastic to have and works pretty well most of the time. When using an adaptive toolpath for roughing the outside contour of the part, the tool plunges down into the stock material a little bit which is very undesirable. 

To solve this, I tried to add a LeadInOut Dressup which does not work, presumably because the first cutting path is vertical and not horizontal like on a Profile operation for example.

The settings changed for the Adaptive toolpath are 'Side' to 'Outside'. Changing from 'Clearing' to 'Profile' does not affect the issue. Enabling 'Force Clearing Inside Out' does not create a reasonable toolpath for roughing the contour of the part (when there is only little stock on the sides).

Using Adaptive with additional LeadIn is better than a Profile operation as that one lacks a step-over parameter or expansion setting (see issue [#10620](https://github.com/FreeCAD/FreeCAD/issues/10620)).

### Full version info

```shell
[code]
OS: macOS 13.0
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * freecad.gears 1.0.0
  * FeedsAndSpeeds 0.5.0
  * Dracula 0.0.5
  * Manipulator 1.5.0
  * fasteners 0.4.63
  * CurvedShapes 1.0.4
  * Help 1.0.3
  * Curves 0.6.13
  * Assembly3 0.12.0
[/code]
```

### Subproject(s) affected?

None

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

davidgilkaufman, 2025-12-12:
> When using an adaptive toolpath for roughing the outside contour of the part, the tool plunges down into the stock material a little bit which is very undesirable.

I've seen this behavior before, and we should fix this. But I don't want to fix it by integrating LeadInOut with Adaptive -- by design, Adaptive is supposed to handle its own entrances and exits. The appropriate fix for this is to ensure that Adaptive does this plunge on the outside of the material.

sliptonic, 2026-02-20:
Changed title to reflect the real underlying problem.