FreeCAD/FreeCAD #10624: [Problem] Mouse navigation: TinkerCAD style has rotation issue in sketcher

Roy-043, 2023-09-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

With Mouse navigation set to TinkerCAD, when you end view rotation in sketch mode the context menu pops up.

Forum topic:
https://forum.freecad.org/viewtopic.php?p=705064#p705120

### Full version info

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34137 (Git)
Build type: Release
Branch: master
Hash: 7ca834acc58841dcf2d66d186db818c8f35f359d
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-11-28:
@Rexbas any interest to fix this ? (And #10263)

Rexbas, 2023-11-28:
Yes, I will take a look.

PR #11841 (2024-01-05): Right click context menu fixes
This PR fixes several right click context menu issues.

The OpenSCAD navigation style did not show the context menu when right clicking. Now it does.
The TinkerCAD navigation style showed the context menu after a fast drag. Now it only shows it when right mouse is released and the view was not dragged.

This PR fixes #10624 where the context menu always popped up after a drag in the sketcher. And the same for the OpenSCAD style but then for panning.

To fix these issues I added two variables: `hasDragged` and `hasPanned`. These are used to detect if the view was  [... cut]
closed 2024-01-05 by chennes