FreeCAD/FreeCAD #10634: [Problem] Draft: Commands can be started after closing all 3D views

Roy-043, 2023-09-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Draft commands can be started after closing all 3D views:
1. Start FreeCAD.
2. Open a document (if required).
3. Switch to the Spreadsheet Workbench.
4. Create a spreadsheet and activate it.
5. Close the 3D view.
6. Switch to the Draft Workbench.
7. Commands like Draft_Line can be started, but result in an error.

### Full version info

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34137 (Git)
Build type: Release
Branch: master
Hash: 7ca834acc58841dcf2d66d186db818c8f35f359d
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2023-09-11:
I can confirm this, but technically it's not wrong. While one of the tabs is still open, even if it's not a 3D view, there is still a document open. The Draft command actually still works, you can enter coordinates by hand, the line will be created.

Should we prevent Draft to work if there is no 3D view open? I am not sure here... What do you think @Roy-043 ?

Roy-043, 2023-09-11:
Multiple documents can be open without a 3D view. It is then unclear to which document new objects will be added.

If we change the `get_3d_view` function to only return the active 3D view (and remove the warning) we can use it in the `IsActive` function:
```
    def IsActive(self):
        return gui_utils.get_3d_view() is not None
```

yorikvanhavre, 2023-09-11:
That's a good idea

Roy-043, 2024-05-18:
Fixed with #14103.

closed 2024-05-18 by Roy-043