FreeCAD/FreeCAD #10639: [Ribbon UI] Missing Short Name for Commands

geolta, 2023-09-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

*Based on the discussion in https://forum.freecad.org/viewtopic.php?t=79235*

For a Ribbon UI, short names for commands are required since they are displayed below the button if it is a large button. For an entry in the menu bar, longer names are okay (e.g. "Create Polyline" for menu and just "Polyline" for the ribbon). Until now, the texts in `menuText` are optimized for classic menu bars and no additional short name is available.

[image]
vs. (although the margins are now maybe too small – but this is an issue not related to the text problem)
[image]

For solving this problem, I see two potential solutions:
- add `shortName` or `ribbonName` property to command
- create additional file (XML, JSON, whatever) where these short names are saved
- more ideas?

Furthermore, the solution should fulfill the following requirements:
- translatable
- working for multiple FreeCAD versions
- the developer is more or less forced to at least think about a short name to avoid a new command is added without being compliant to a ribbon UI

In my opinion (but my experience with FreeCAD code base is quite limited) these requirements are pointing towards extending the `Command` object. What are you thinking about this?

*(I want to emphasize, that solving this problem is a prerequisite for implementing a good ribbon UI, regardless of implementing it in Python or C++.)*

### Full version info

```shell
OS: Fedora Linux 38 (KDE Plasma) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.2.Unknown
Build type: Unknown
Python 3.11.1, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2023-09-10:
This does not belong here, it should moved to the ribbon ui add-on tracker.

geolta, 2023-09-12:
I do think that it belongs here since the changes apply to FreeCAD master code, not to some addon.

luzpaz, 2023-09-17:
I like the idea of `shortname`.

grubertechnic, 2023-09-25:
> I do think that it belongs here since the changes apply to FreeCAD master code, not to some addon.

So you want to change something in the master for an addon, make little sense to me.

80avin, 2023-12-30:
I second the idea of doing the change in Freecad than the plugin.
A few points in support of the issue which aren't mentioned previously could be:

Why such names are generally not good for UI elements ? :
* From the perspective of UI design, "Create" in "Create Polyline" is just redundant and wastes unnecessary space while creating more text for the user to scan. It would've been fine if there were only one such command, but when multiple commands are named "Create Polyline", "Create Rectangle", etc, this issue becomes more prominent.
* Particularly in the "Constraints" group ( and similar ), when the command is already defined to be a part of the "Constraints" group, it is completely unnecessary to prefix every command with "Constraint " text

Considering that such names are generally not good for UI elements, Why should this be fixed in FreeCAD and not the plugin ? :
* FreeCAD code is the source of this naming pattern, and if the pattern is wrong, it should be fixed at the source.
* Asking the UI plugin to have the "Create Polyline" to "Polyline" mapping means that
   * All such UI plugins need to have such mapping.
   * They require this mapping for all supported languages, which would slow down the plugin development or affect the quality of an entire class of such plugins.
   * Any new command added by any plugin or FreeCAD update can break the UI plugin which makes this issue an issue of inter-compatibility of multiple plugins, suggesting that it should be fixed in FreeCAD itself.

###### Feel free to correct me.

maxwxyz, 2024-08-30:
I think instead of adding something for one addon we should discuss options of changing the UI in main.