FreeCAD/FreeCAD #10641: [Ribbon UI] Missing different sizes for icons

geolta, 2023-09-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

*Based on the discussion in https://forum.freecad.org/viewtopic.php?t=79235*

Currently, all icons have the same size. But a ribbon UI is based on the idea of having different sizes for icons. When just scaling up the icons you get different line thicknesses what looks awkward:

[image]
vs. 
[image]

Two issues arise from this:
- Up to my knowledge, there is no "easy" solution to scale the icons without changing the line thickness (also scaling without changing the stroke width doesn't work due to the inner "Highlight Color" (according to the Artwork Guidelines). Maybe someone knows some AI solution or so? But since only a couple of icons would need this, this could be done manually as well without too much effort.
- Change the method that is loading the icon to get the correct size. This would also require some change to the current way how icons are stored (information about their size would be needed) (although I don't know how this is currently handled).

What do you think about possible solutions for implementing this? I have to dig into the icon code first before I can add anything valuable to the discussion ;(

### Full version info

```shell
OS: Fedora Linux 38 (KDE Plasma) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.2.Unknown
Build type: Unknown
Python 3.11.1, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

NXTler, 2023-09-10:
This

grubertechnic, 2023-09-10:
> This

does not belong here, it should moved to the ribbon ui add-on tracker.

FlachyJoe, 2023-09-10:
This can be solved by using `vector-effect="non-scaling-stroke"` in SVG icons. See https://developer.mozilla.org/en-US/docs/Web/SVG/Attribute/vector-effect

geolta, 2023-09-12:
> This can be solved by using `vector-effect="non-scaling-stroke"` in SVG icons. See https://developer.mozilla.org/en-US/docs/Web/SVG/Attribute/vector-effect

Unfortunately this doesn't work since the position of the strokes is not updated. See this two examples here (scaled up by factor 3):
[image]
[image]

FlachyJoe, 2023-09-12:
SVGs certainly need a tweak. You maybe can find more at https://muffinman.io/blog/svg-non-scaling-circle-and-rectangle/

grubertechnic, 2023-09-25:
still, a ribbon addon issue, need to be moved.

luzpaz, 2023-09-29:
Where is the ribbon ui repo ?

adrianinsaval, 2023-10-09:
open this in the appropriate repo please

luzpaz, 2023-10-11:
It's at https://github.com/geolta/FreeCAD-Ribbon

luzpaz, 2023-10-11:
@adrianinsaval the question is not answered.

luzpaz, 2023-10-11:
Looks like there is a related 'icon size' ticket in #10694

adrianinsaval, 2023-10-12:
> @adrianinsaval the question is not answered.

the issue tracker is not for help requests. This is not a FreeCAD issue.

adrianinsaval, 2023-10-12:
> Looks like there is a related 'icon size' ticket in #10694

not really, that one is about the lack of padding between icons in the new themes by mistermaker, here the user just doesn't like how the icons look when upscaled, which is very subjective (I thing it looks perfectly fine for example).

If you rework the icons so they look the same but scale on a more satisfactory way for you we can accept it, but there's really no point in keeping this in the tracker, it does not result in any FreeCAD problem so there's no action for us to take.

luzpaz, 2023-10-12:
@adrianinsaval to @FlachyJoe's point though...?

> SVGs certainly need a tweak. You maybe can find more at https://muffinman.io/blog/svg-non-scaling-circle-and-rectangle/

ref: https://github.com/FreeCAD/FreeCAD/issues/10641#issuecomment-1715553047

adrianinsaval, 2023-10-13:
not a problem for our usecase, tweaking would only required to make them scale as geolta `wishes`. The idea that that would be better is very subjective, if he's willing to personally rework all icons so they can work like that we could accept a PR but it's not worth changing for us.

adrianinsaval, 2023-10-16:
closing again as there's no reasonable action to be taken on our side here.

closed 2023-10-16 by adrianinsaval

mclegrand, 2023-11-19:
> Unfortunately this doesn't work since the position of the strokes is not updated. See this two examples here (scaled up by factor 3): [image] [image]

Isn't that because you are trying to use stroke for objects you *want* to scale, and which could be a fill ? Using fills for parts you want to scale and strokes with non-scaling stroke for your borders should do what you want ?