FreeCAD/FreeCAD #10646: [Problem] Sketch after merge not editable

PLChris, 2023-09-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

- Open the attached file
- select Sketch001
- select Sketch (this sequence is important!)
- merge sketches
- the result is Sketch003 which cannot be edited without using ValidateSketch and remove all constraints

### Full version info

```shell
OS: macOS 13.0
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34137 (Git)
Build type: Release
Branch: master
Hash: 7ca834acc58841dcf2d66d186db818c8f35f359d
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * FC_SU
  * freecad.gears 1.0.0
  * FeedsAndSpeeds 0.5.0
  * dxf-library
  * fcgear 1.0.0
  * fasteners 0.4.59
  * sheetmetal 0.2.62
  * ExplodedAssembly
  * Curves 0.6.8
```

### Subproject(s) affected?

Sketcher

### Anything else?

rename the extension of the file to FCStd
[mergeSketchesFail.zip](https://github.com/FreeCAD/FreeCAD/files/12570895/mergeSketchesFail.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-09:
It seems to have problems with the constraints to the external geometry, but only in this selection order.
@FlachyJoe would you be able to look at this?

FlachyJoe, 2024-03-09:
Place to start:
https://github.com/FreeCAD/FreeCAD/blob/5b5be480a3a14963cbdefe2e7a3923f9136af6a9/src/Mod/Sketcher/App/SketchObject.cpp#L8876-L8892

luzpaz, 2024-03-15:
Can someone synthesize a succinct title for this ticket based on @FlachyJoe's diagnosis ?

ornea, 2025-02-18:
I have this issue.  - merged sketch with external geometry breaks
My laborious work around was to use a spreadsheet and create a alias for every single dimension.  This is way less cool than referencing external geometry

FEA-eng, 2025-09-17:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/16 20:38:23
Build type: Release
Branch: main
Hash: bac49e465908071e7400a547eb813bc5017ae7dc
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```