FreeCAD/FreeCAD #10651: [Problem] Sketcher_Dimension doesn't account for overconstraint in coordinate dimensions

FEA-eng, 2023-09-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

@paddleThe new Sketcher_Dimension tool could be improved by making it adjust the coordinate dimensions (and possibly other types of dimensions) to avoid potential overconstraints like this:

[image]

Of course, in this case, only the vertical distance should be defined.

### Full version info

```shell
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34223 (Git)
Build type: Release
Branch: master
Hash: 577004c1eba55641e92ea085fa6a078d36f49a06
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2023-09-12:
We could check if the constraint is overconstraining everytime and skip the constraints that would be conflicting. 
In the case you screenshot it would make sense. But I feat that in some cases it would atually be confusing. Say you want to administer a distance constraint to a line. But can't because it's conflicting. If it doesn't even appear it would be confusing. Whereas if it appears but you see it's conflicting, then you can still create it, find the conflicting constraint you want to remove.
Thoughts?

BanceFC, 2023-09-13:
Similarly, it is not possible to simply dismiss either 'X' or 'Y' when trying to constrain a circle in one direction only.

FEA-eng, 2023-09-13:
> Thoughts?

Right, it's not so straightforward. Maybe the tool could be more similar to smart dimensioning in commercial CAD software where selecting a point doesn't create those coordinate dimensions by default but waits for the user to select another reference. In this case, selecting the center of the sketch should only create the vertical dimension.

Those coordinate dimensions are a nice addition (no need to select more references in some cases) but maybe they could be disabled by default and enabled only when a specific key is pressed.

PaddleStroke, 2023-09-13:
Yes perhaps the 'lock' constraint mode on point is overkill. Though it can be useful in some cases. As the lock contraint is sometime useful I guess.
I have two ideas : 
- Add an empty mode first for 1 point. So there would be no constraints when you select a point until you click 'M' key, then it goes to lock.
- Make a setting to disable Lock.

BanceFC, 2023-09-13:
What if I want to constrain horizontal to one point and vertical to another?

howie-j, 2023-09-13:
>     * Add an empty mode first for 1 point. So there would be no constraints when you select a point until you click 'M' key, then it goes to lock.

This is the way imo. That also fixes the issue that it's hard to select the second point because the lock dimensions get in the way. And no need for yet another preference :)

howie-j, 2023-09-13:
> What if I want to constrain horizontal to one point and vertical to another?

Then you do just that? Nothing stopping you from doing that now.

BanceFC, 2023-09-13:
OK, I see now that takes us back to the original problem.
But is it possible to simply dismiss one of the dimensional constraints, if chosen or if there is already a constraint as shown above?

howie-j, 2023-09-13:
I swapped the order of the constraint options, so that the default for one point selected is autodimension, and you press M for lock. Is this a possible solution @FEA-eng @BanceFC?

[fix.webm](https://github.com/FreeCAD/FreeCAD/assets/82042580/50958e71-05d1-4a9f-a256-a8750b38faef)

It does not catch potential overconstraints, but i'm not sure if dimension should handle all possible overconstraints or if we should just let "auto remove redundant constraints" handle that..

FEA-eng, 2023-09-14:
@howie-j It's a very good solution, in my opinion :-)

BanceFC, 2023-09-14:
Looks like it'll work.

PR #10679 (2023-09-18): [Sketcher] Change Sketcher_Dimension 1point constraint order
Proposal to mitigate/fix #10651

Changes Sketcher_Dimension constraint options for 1point from [lock, distance] to [distance, lock]

@PaddleStroke
commit 12913dae (2023-09-18): fixes #10651
closed 2023-09-18 by chennes