FreeCAD/FreeCAD #10661: [Assembly] Name for joints

PaddleStroke, 2023-09-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

We need to decide on a name for the objects that are going to join parts.
It can be Constraint, it can be Mate, it can be Joint.

Currently I used 'Joint' and 'joint-connector' in my [draft PR](https://github.com/FreeCAD/FreeCAD/pull/10611) but this is of course open to discussion.
Joint because this is what Dr Koh is using in his solver.

There is going to be different type of joints : Fixed, revolute, cylindrical, slider, ball, parallel, coplanar, tangent.

So do you think that this naming is good : 

> Fixed Joint
> Revolute Joint
> Cylindrical Joint
> Slider Joint
> Ball Joint
> Parallel Joint
> Coplanar Joint
> Tangent Joint

Or do you think something else is better suited?

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

qewer33, 2023-09-12:
I mentioned it in the other issue but it applies here as well, it would be great have visual examples and text descriptions of these joints in order to think of proper names

PaddleStroke, 2023-09-12:
@qewer33 Here's a video from onshape where you can see a very similar workflow : https://www.onshape.com/en/resource-center/videos/build-your-first-assembly

tobiasfalk, 2023-09-12:
Joint sounds good.
Constraint could be confused with the sketch constraints.
Mate also could work but personally I do not "feel" it, it does not sound right.
Additionally it would be clever to use the same name as the solver, this makes working with it easier.

WandererFan, 2023-09-12:
In the Open Assembly Model, I believe what you are calling a joint is an AssemblyFeatureAssociation which describes the relationship between 1 or more AssemblyFeatures.   But those names are data model names that aren't very end user friendly and are better suited to the internals than to the UI.

[image]

[image]

closed 2023-09-20 by PaddleStroke

maxwxyz, 2023-09-27:
What about "connectors", "link" or "relation"? 

I'm curious if there is a simple description of what these joints do? Is is possible to add offsets to some joints (distance, rotation)?
Is there an angle joint to constrain between two faces or how would I realize an assembly where I only want to constrain an angle?