FreeCAD/FreeCAD #10662: [Assembly] Icons for joints

PaddleStroke, 2023-09-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

We need some good icons in-line with freecad style to cover the different types of joints : 

> Fixed Joint
> Revolute Joint
> Cylindrical Joint
> Slider Joint
> Ball Joint
> Parallel Joint
> Coplanar Joint
> Tangent Joint

Anyone with skills wants to jump in?

Draft PR on joints : https://github.com/FreeCAD/FreeCAD/pull/10611

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

qewer33, 2023-09-12:
It would be great have some sort of "visual examples" of these joints to replicate in the icon style

qewer33, 2023-09-12:
I've started working on these with inspiration taken from OnShape's icons. Let me know how they look:

[image]

The idea is to have a "base" object (Cylinder in this case) and then red sketcher-like "constraint" icons on top of it.

Here is OnShape's icons:

[image]

onekk, 2023-09-12:
Maybe this could be a decent overview, of different incarnations, as mimicking other software should not be encouraged, as it could incurr in copyright infringiments if they are "too similar".

https://blog.3dcs.com/cad-joints-and-constraints-how-to-use-catia-nx-and-creo-joints-and-constraints-tolerance-analysis

Maybe simply changing the shape, as example using an exagon instead of a cylinder should make enough differences to not recall immediatly the "inspiration icons".

IMHO as usual.

Regards

Carlo D.

kadet1090, 2023-09-12:
> [image]

For the Slider icon I'd personally put arrows under the cylinder in isometric view, as in most cases this join in case of cylindrical object will be applied to base of the cylinder.

PaddleStroke, 2023-09-12:
Note slider is up-down along one axis. So the double arrows would rather be for coplanar.
Also fixed, I think there might be an ambiguity here as we will have 2 kind of fixed.
- The first one is going to be a 'fix part' that will lock a part in place.
- Then there is what I called 'Fixed Joint' that is a joint between 2 parts that leaves no DOF. So it fixes two parts together.

That's maybe why onshape is using the term 'fastened' for this joint.

qewer33, 2023-09-12:
Here's what I got currently:

[image]

I have no idea what to do for cylinder joint as that just seems to be revolute but with slide motion, maybe same as revolute but with an up-down arrow?

qewer33, 2023-09-12:
> Also fixed, I think there might be an ambiguity here as we will have 2 kind of fixed.
>
>     * The first one is going to be a 'fix part' that will lock a part in place.
>
>     * Then there is what I [...]

I don't think "fastened" is that great of a name for a joint like this either but yeah fixed might only imply fixing a part in place in 3D space, like how other Assembly WBs (Asm3 and A2+) work.

qewer33, 2023-09-12:
> For the Slider icon I'd personally put arrows under the cylinder in isometric view, as in most cases this join in case of cylindrical object will be applied to base of the cylinder.

I'm not sure I understand, can you elaborate?

kadet1090, 2023-09-12:
> > For the Slider icon I'd personally put arrows under the cylinder in isometric view, as in most cases this join in case of cylindrical object will be applied to base of the cylinder.
> 
> I'm not sure I understand, can you elaborate?

Nvm, I thought that this is icon for Planar joint, now everything is clear!

qewer33, 2023-09-12:
Last iteration (still not happy with Cylindrical though):

[image]

@PaddleStroke zip file: [zJoint_Icons.zip](https://github.com/FreeCAD/FreeCAD/files/12589649/zJoint_Icons.zip)

It might also looks nice to make 3D-ish icons similar to ones in the link sent by @onekk but I don't have the time to create those, would be nice to see someone else trying that idea out.

qewer33, 2023-09-12:
Made an alternate set with tweaks:

[image]

Files: [zJoint_Icons_Alt.zip](https://github.com/FreeCAD/FreeCAD/files/12590232/zJoint_Icons_Alt.zip)

tobiasfalk, 2023-09-12:
I like the alternative version, but I would make the blue cylinder also look out on the bottom. So that the yellow one is more of a ring on a cylinder. This goes for slide, revolt and cylindrical joint.

qewer33, 2023-09-13:
> I like the alternative version, but I would make the blue cylinder also look out on the bottom. So that the yellow one is more of a ring on a cylinder. This goes for slide, revolt and cylindrical joint.

Good idea, here's how they look:

[image]

Files: [zJoint_Icons_Alt.zip](https://github.com/FreeCAD/FreeCAD/files/12593848/zJoint_Icons_Alt.zip)

PaddleStroke, 2023-09-13:
Haha damn I was just doing that ! Didn't thought you'd be up so early doing icons :)
[image]
Here I offer alternate very similar to the one you just posted.
2 small differences : 
- I removed the red lines of the slider / cylindrical. I think they don't bring much and clutter the icon.
- The arrows are a tiny bit more curved.
Files : [Joint_Revolute_Alt.zip](https://github.com/FreeCAD/FreeCAD/files/12593883/Joint_Revolute_Alt.zip)

qewer33, 2023-09-13:
> Haha damn I was just doing that ! Didn't thought you'd be up so early doing icons :)

I had a little free-time in my hands so I thought I'd fill it up :smile: 

>     * I removed the red lines of the slider / cylindrical. I think they don't bring much and clutter the icon.
> 
>     * The arrows are a tiny bit more curved.
> 
>     * The ring is a bit wider

Yours look better. And on second thought, I agree that the red parallel lines are better removed.

qewer33, 2023-09-13:
@PaddleStroke I've rotated and adjusted them to give an isometric 3D effect as an experiment:

[image]

Files: [Joint_Revolute_Alt.zip](https://github.com/FreeCAD/FreeCAD/files/12599555/Joint_Revolute_Alt.zip)

PaddleStroke, 2023-09-13:
I like those !
Few last remarks : 
- Ball : It doesn't look much like a ball. I think the arrows are a bit too thick, and curving them a little bit more would look better. (so that the center of the arrows would be further away from the ball center)
- Tangent, rather than a blue line we could use a blue cylinder. And perhaps an isometric view too.

tobiasfalk, 2023-09-13:
> @PaddleStroke I've rotated and adjusted them to give an isometric 3D effect as an experiment:
> 
> [image]
> 
> Files: [Joint_Revolute_Alt.zip](https://github.com/FreeCAD/FreeCAD/files/12599555/Joint_Revolute_Alt.zip)
> 

In Cylindrical, I would put the red arrow for sliding g on top of the ring and not below

qewer33, 2023-09-13:
>     * Tangent, rather than a blue line we could use a blue cylinder. And perhaps an isometric view too.

Something like this seems to look better, also updated the ball and cylinder:

[image]

Files: [zJoint_Icons_Alt.zip](https://github.com/FreeCAD/FreeCAD/files/12600036/zJoint_Icons_Alt.zip)

PaddleStroke, 2023-09-13:
@qewer33 do you want to open a PR to introduce those icons ?
If so please rename them to follwo the proper naming convention. See the assembly.qrc with the names : 

```
<RCC>
    <qresource prefix="/">
        <file>icons/Assembly_InsertLink.svg</file>
        <file>icons/preferences-assembly.svg</file>
        <file>panels/TaskAssemblyInsertLink.ui</file>
        <file>preferences/Assembly.ui</file>
        <file>icons/Assembly_CreateJointBall.svg</file>
        <file>icons/Assembly_CreateJointCylindrical.svg</file>
        <file>icons/Assembly_CreateJointFixed.svg</file>
        <file>icons/Assembly_CreateJointParallel.svg</file>
        <file>icons/Assembly_CreateJointPlanar.svg</file>
        <file>icons/Assembly_CreateJointRevolute.svg</file>
        <file>icons/Assembly_CreateJointSlider.svg</file>
        <file>icons/Assembly_CreateJointTangent.svg</file>
    </qresource>
</RCC>
```

qewer33, 2023-09-13:
> @qewer33 do you want to open a PR to introduce those icons ? If so please rename them to follwo the proper naming convention. See the assembly.qrc with the names :

Sure, I can. I need to add them to `Mod/Assembly/Gui/Resources/icons` like in your draft PR right?

PaddleStroke, 2023-09-13:
Yes that's it

qewer33, 2023-09-13:
@PaddleStroke PR opened,you can also mention me on the forum or Discord server if/when there are more icon needs for the Assembly WB. Would be glad to help!

closed 2023-09-14 by sliptonic

MisterMakerNL, 2023-09-20:
Icons looke awesome! well done!

maxwxyz, 2023-09-27:
I really like the new icons. Why do we need a separate joint for the combination of slider and revolute (=cylindrical)?
These icons are very similar as well. Would it be possible to make the sliding icon a sliding box or rectangle which is already visually not possible to rotate? Currently all three icons are cylinders.
For tangent I would prefer a plane on a cylinder or ball. Based on the icons I do not understand the difference between tangent and ball joint.