FreeCAD/FreeCAD #10691: [Problem] TechDraw text too big after update to 0.21

FesixGermany, 2023-09-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After updating from 0.20 to 0.21 the text in my existing drawings is too big.
I always used 3mm text height for dimensions, now in 0.21 I need to reduce the height to about 2.2mm for everything to fit again.
Also annotations went bigger, for example using 4mm needs to be reduced to 3mm, positions also might have shiftet a little bit?

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2023-09-15:
Text in V0.20 was undersized relative to the nominal height.

If you measure the text, you'll find that it matches the nominal height in v0.21.

There is a macro in this issue report #10413 that can adjust text to match v0.20.

FesixGermany, 2023-09-15:
Thank you very much, didn't find that issue when I searched for that problem, not very happy about drawings getting somewhat torn apart when opened with newer versions but at least that macro seems to work fine.

WandererFan, 2023-09-15:
Yeah, looking back, I should have put in a switch to control whether to use the old (incorrect) sizes or the new (correct) sizes. 

The macro was a bit of a rush job, glad it is working for you.   Depending on the font you are using you may need to adjust the scaling factor.