FreeCAD/FreeCAD #10694: [Problem] command groups in toolbars are too small in Dark/ProDark.

PaddleStroke, 2023-09-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On linux : 
[image]
On windows : 
[image]

It is now very tedious to select tools. Even worse on windows.

In the dark theme it's even worse : 
[image]

I think this has been recently introduced. @MisterMakerNL is it part of your rework on the stylesheet?
If so can this be reverted ?

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34301 +3 (Git)
Build type: Release
Branch: move_box_datumLabel
Hash: e3bc1ce92429bf5b25b5c5f52578fd1853a8c86f
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk , OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.60k
  * Assembly3 0.12.0
  * Assembly4 0.50.2
  * Curves 0.6.8
  * ExtremeProLight 1.5.0
  * FreeCAD - Shortcut.lnk
  * FreeCAD_FastenersWB-master 0.4.56
  * Ondsel 2023.8.23.02
  * ProDarkThemePreferencePack 1.0.0
  * sheetmetal 0.2.63
[/code]
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MisterMakerNL, 2023-09-15:
Issue here is the vast amount of icons freecad has and the old solution losing lots of screenspace. And the layout of the toolbars are not aligned.
Furter this option has been tested in linkstage without any complaints. I personally think it works great.
But nevertheless we can try a optimize it. I know there is an issue with icon size and the size if the pull down icon.

kadet1090, 2023-09-16:
Personally I really like this behavior because when the toolbar is placed on the side it does not waste space and it nicely align icons. But a little bit more clicking space would be welcome.

[image]

adrianinsaval, 2023-10-12:
I talked of this before too, at least a minimum amount of padding should be present. No need to go exactly to what we had before, but at least a middle ground. In any case if you really want a stylesheet like this it could be provided as a compact variant.

closed 2024-07-02 by PaddleStroke