FreeCAD/FreeCAD #10698: software-developer-centric error messages[Problem]

mbrown85209, 2023-09-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Ok, I don't have a lot of traction in FreeCAD yet, but as a newbe one of the first things I noticed was that error messages which appear in the Report View are too cryptic to be discernable to anyone who is not a FreeCAD software developer who happens to have a debugger open, etc.

All error messages that regular users see should say something they will understand and can act on -- error messages with stack-trace and variable info should be reserved for log files.  

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kadet1090, 2023-09-15:
Personally I think that all error messages should be reviewed before 1.0 release. Why? Because 1.0 is a really "big" release for every software, so FreeCAD will probably get a lot of coverage on the web, and we should aim to not scare off people with messages that are meaningless for normal users. For example all references to things like `BRepAPI_MakeOffset` should be removed from user facing messages. If we cannot provide more information in the message because it simply is not provided by the OCC we can always replace it with some more general stuff like.

```
BRepAPI_MakeFillet failed -> Cannot make fillet, please check underlying geometry and parameters.
```

It may not provide more information to the user, but at least it is something that user can understand because it is in plain english.

luzpaz, 2023-09-16:
Any volunteers to compile a list of the esoteric error messages somewhere ?

kadet1090, 2023-09-16:
I could try to create some script that'd search for code symbols across strings to ease the job.

mbrown85209, 2023-09-16:
As a retired software developer myself, I found that truly effective software that includes user interfaces or user-generated inputs needed to be “idiot-proofed” to about the same degree as unit testing itself.  For example, popular YouTube gaming channels like to exploit loopholes where users do something bizarre to cause the software to misbehave with spectacular results (almost as if they were intentional Easter Eggs). I always roll my eyes at those because it reveals naïve, or at least lackadaisical error checking.

 

Although I never worked with Python, I wrote a LOT of Java code and acknowledged my lazy workarounds to effective error checking when I simply “threw” Exceptions up the stack from one method to another, until somewhere I finally trapped them all to print the last few nodes of the stack-trace as a note to myself for a clue where to go debug the problem.  And it seems this is what is happening in FreeCAD.

 

As a newbie to FreeCAD, I see these “stack-trace” error messages just about everywhere I go in the system while I am trying to learn how to use it.  It seems that the “user-unfriendly” error messages that this issue refers to are rather ubiquitous, which means to me that user inputs are not well-scrutinized.  So, I imagine the solution would be equally ubiquitous, not just a list of the more common ones.

 

If I was in a training class and I encountered an “esoteric” error message, I would turn to my instructor and ask what that meant and what I did wrong, To which they would respond with an explanation and corrective action; To which I would complain that the error message should have given me that information in the first place.

kadet1090, 2025-01-23:
Still valid - @maxwxyz would be a good one to mark as not getting stale.