FreeCAD/FreeCAD #10701: [Problem] Std_Part cannot be exported to IGES

Roy-043, 2023-09-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

A Std_Part cannot be exported to IGES. A Std_Part can however be exported to other formats (STL, STEP).

Steps to reproduce:
1. Create a Part_Box.
2. Create a Std_Part.
3. Put the box in the part.
4. Select only the part.
5. Call Std_Export and select the IGES format.
6. Open a new FreeCAD document.
7. Import the created file.
8. No object is created, so apparently the created file does not contain any geometry.

Forum topic:
https://forum.freecad.org/viewtopic.php?t=81209

### Full version info

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34223 (Git)
Build type: Release
Branch: master
Hash: 577004c1eba55641e92ea085fa6a078d36f49a06
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2025-01-27:
This is still an issue.
```
OS: Windows 8 build 9600
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

FEA-eng, 2025-06-08:
Still relevant