FreeCAD/FreeCAD #10702: Path Simulator Speed

mbrown85209, 2023-09-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Presently, the functionality of the Path Simulator seems to be based on the interests of the developer rather than the interests of the user.  Perhaps developers want to confirm that the software is running properly, but users want to confirm that their programs will ultimately cause the CNC machines to behave and make parts properly.

I suggest altering the functionality of the simulator to at least introduce an option to control the feed rate of the tool versus the rate at which it reads lines of gcode.  So, the Speed slider actually affects the speed of the tool as it proceeds along the path by scaling the feed rate specified in the operations, and at a faster or slower rate.

That is, after all, why the user needs to simulate physical behavior - to look for unexpected activity, like incorrectly specified speeds and feeds, fixture, clamp, and tool holder collisions, incorrect tooling, gouging, inefficient or incorrect tool paths, etc.

A CNC machine operator NEVER runs a new program on a machine without assuring it isn't going do something bad or poorly, Hardware mistakes are simply too expensive and dangerous to happen and must be avoided with all effort.  CAM programmers need simulators to allow them to do no less.

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

Path

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-09-16:
There are efforts to port the path simulator code out to camotics

[image]

Not sure what the status on it ATM. Lots of attention and resources are going in to fix toponaming. Maybe @sliptonic can weigh-in on camotics (if he has a moment?)

sliptonic, 2024-05-07:
Work by shaise is likely replacing the old simulator.  I'm closing this issue in anticipation of that.

closed 2024-05-07 by sliptonic

luzpaz, 2024-05-08:
1. https://github.com/FreeCAD/FreeCAD/pull/13884
2. https://forum.freecad.org/viewtopic.php?t=87174