FreeCAD/FreeCAD #10704: [Feature Request] Midpoint constraint in sketcher

dyylanhammond, 2023-09-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Freecad has snap to midpoint in sketcher, but not a midpoint constraint. The only way I've found to do it currently is to use the symmetry constraint, which is slow and unintuitive. Adding an automatic midpoint constraint would make sketching faster and easier.

### Full version info

```shell
[code]
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * PieMenu 1.2.7
[/code]
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

qewer33, 2023-09-16:
I don't think the symmetry constraint is "slow and unintuitive". You can just select a line and a point and then apply symmetry. It acts exactly like how a midpoint constraint would act like. Would be nice to make that automatic for the midpoint snap though yes.

dyylanhammond, 2023-09-16:
I didn't know it could be used with a point and line, that functionality isn't explained in the tooltip and is barely explained in the [wiki](https://wiki.freecad.org/Sketcher_ConstrainSymmetric/en). I was using it as described in the tooltip. A better tooltip and being applied automatically would help.

luzpaz, 2023-09-17:
How would a more ideal tooltip look like ?

howie-j, 2023-09-17:
How about something like this?

Currently:
[image]

Proposal:
[image]

I can create a PR when we reach agreement.

dyylanhammond, 2023-09-17:
Yeah, that would help.

kadet1090, 2023-09-17:
TIL that symmetry can affect also symmetry point. I however think that this solution has 2 problems at least:
1. I would not even think to look under the symmetry for doing midpoint stuff. This is technically symmetry constraint, but the user does not think in that way.
2. Symmetry works only for lines, and other edge types also have quite well defined midpoints. 
So personally I'd prefer to have "Midpoint" constraint available on the toolbar, as it is better aligned with what user expects.

dyylanhammond, 2023-09-17:
Exactly my original point.

howie-j, 2023-09-17:
Yes i thought about that also, however adding yet another icon to the toolbar might not be ideal.

What about renaming the constraint from just _symmetry_ to _symmetry/midpoint_ or similar, and changing the icon to something more suited to aid discoverability of this function?

chennes, 2023-09-17:
I'm looping @PaddleStroke in on this discussion to see if the recent work he has been doing on the auto constraints might give him some insight into the best approach here.

PaddleStroke, 2023-09-18:
Imo just improving the tooltip to details the different cases is sufficient.
And yes it has been asked several time for a symmetric constraint to be created on middle point snap. I have not touched the auto-constraint code yet so I'm not sure how to implement it but I can have a look.

kadet1090, 2023-09-18:
> Imo just improving the tooltip to details the different cases is sufficient.

And what about midpoint of an arc? This cannot be solved by symmetry. We could of course update symmetry constraint to support that case, and later for bsplines (not sure if feasible, or even useful) but I feel that this constraint would try to do too many things.

For midpoint and symmetry are two different constraints, because the user expectations are different. It just happens that in the line case midpoint can also be achieved using symmetry but this is workaround IMHO.

But if we decide that another constraint is not required.it should at least change name and tooltip like it was suggested. No one reads tooltip of tools which do not seem even considerable for the

PaddleStroke, 2023-09-18:
@kadet1090 Ok I can see the use of an independant middle constraint now as indeed it would work both on line and arcs.

But then this is much more work as it would need to create a new type of constraint. I'm not even sure where to begin. @FlachyJoe recently implemented a new constraint (circle-arcs distances). Perhaps this is something you'd be interested in as well?

FlachyJoe, 2023-09-18:
I add this to my [todo list](https://github.com/users/FlachyJoe/projects/1/views/1).

NXTler, 2023-10-16:
Solid Edge and Solid Works do this by adding a midpoint to the line itself. You can attach other geometries to it like you do regular. Here is a Screenshot I found online:
[image]

It only becomes visible when hovering over it, so the sketch doesn't get too cluttered.
I think this is a very intuitive solution that is easy to understand for the user. What do you think?

wertig0n, 2024-02-03:
I would like to +1 this FR. To be clear, this should be fairly easy to do, since the constraints are already in place - it is more of a display problem. I think the easiest way to achieve this is to use the tools that already exist and hide the complex geometry.

How *I* would like it to work, if possible just get the midpoint to show up when hovering over an arc or line. An easier target is the simple right click and "Add midpoint" option, and perhaps a hotkey coupled to this. This should add something like this (but "Constrained to Object" icon is a midpoint icon instead):

[image]

In the background, however, the following constraints are applied:

[image]

Seems easy enough to do, but if FreeCAD internally does not allow for this, weeeell...

MisterMakerNL, 2024-02-23:
I see some options for improvements here, but not really high priority in my books.
I had issues with this too when starting Freecad, but after some time I kinda started liking it. And I kinda prefere it even above Solidworks.
Reason is that it is always clear it is a midpoint and it is also clear that it is a midpoint between 2 other points.
In Solidworks this can be an issue especially if you have 2 lines on top of each other.

One option for improvement would be to make it more smart, like you click on a line and it could just add a midpoint to the line. You click on 2 lines it could add a center-line. If you click 2 points again a center point. 
It's a bit comber-sum to place a point and then do the thing, while the thing can also place the point.

PR #13147 (2024-04-01): Sketcher: Add line mid-point auto-constraint
Add a line mid-point (symmetric) auto constraint when creating geometries.
[image]
Refactor a little bit of code at the same time (for loop, auto)

Fixes #10704
Fix one part of #11414
Also mentioned in #6240
@pierreporte
closed 2024-04-01 by chennes