FreeCAD/FreeCAD #10705: [Problem] FEM workbench needs an official maintainer or someone engaged until one is found

luzpaz, 2023-09-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

bernd, donovaly and others who have occupied the place of FEM maintainer have stepped down. There are several pending PRs and open issues that holding back the FreeCAD FEM community. An example below shows a PR that is being neglected because attention and resources are being directed elsewhere:  

Per @Syres916 in https://github.com/FreeCAD/FreeCAD/issues/10437#issuecomment-1721535867

> I've requested another maintainer to give the once over the PR (#10437) but I am loosing in faith in the whole process. There's been not one piece of real feedback negative or positive regarding the actual code. The longer it goes on the more chance of significant conflicts that then have to be resolved.

So perhaps we can come to an agreement of an interim dev that is willing to take on maintainership till a more longterm solution is found? 

### Full version info

```shell
n/a
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-09-17:
Issues needing attention:

- [ ] #10437 (Pending PR)
- [ ] #5966 (a new dev needs orientation on how to go about solving issue)

FEA-eng, 2023-09-18:
@luzpaz I would add this one too: #9477

There's also #10439 but it's handled by the same PR as #10437.

luzpaz, 2023-09-20:
Looks like 2 issues were fixed/merged, nice!  
Added #10569

FEA-eng, 2023-09-20:
We could also add this bigger one: #7716

yorikvanhavre, 2023-09-21:
There are many possibilities here. The ideal of course would be that someone with enough time and knowledge of FEM would step up to take the role. But this might take time to happen.

I would rather suggest that people interested in FEM join the [merge meetings on mondays](https://www.freecad.org/events.php) and help people to review and decide over FEM pull requests. We found out in these meetings that if enough people are there to give their opinion on a particular PR, it helps a lot to go forward and merge

FEA-eng, 2023-09-21:
> I would rather suggest that people interested in FEM join the [merge meetings on mondays](https://www.freecad.org/events.php) and help people to review and decide over FEM pull requests.

That's definitely a good idea and I'll try to join those meetings when there's some FEM PR waiting to be merged but the biggest issue is that in some cases we are stuck without the help of someone familiar with the FEM source code. Like with this PR: #10569

yorikvanhavre, 2023-09-22:
If there is nobody around who is knowledgeable about the FEM source code, we'll look and try to find out ourselves ;)

marioalexis84, 2023-09-24:
If it helps, I can help with pull requests related to FEM.

closed 2025-01-26 by luzpaz