FreeCAD/FreeCAD #10707: When the placement is changed in a body, it will fail in any boolean operations with other bodies.

chopsywa, 2023-09-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I was creating a laminate with multiple cloned bodies in part design. Each body needed to be stacked against the prior, so I set the x axis on each subsequent clone. All operations seemed to work fine moving forwards, except boolean operations. They did all sorts of random things. The workaround was to change the placement variations in the clone objects themselves. 

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Australia (en_AU)
Installed mods: 
  * Assembly4 0.50.2
  * Curves 0.6.13
[/code]
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-09-19:
You need to post an example file and steps to reproduce.

luzpaz, 2023-09-23:
@chopsywa :point_up_2:

chopsywa, 2023-09-24:
OK. Simple steps to reproduce.
Create new project.
Create a body. Create a sketch on the XY plane.
Draw a 25mmx25mm square with bottom left corner on the origin.
Pad it to a 25mm cube.
Go into placement and move the body =10mm Y and -10mm Z
Create a new body. Create a sketch on the XY plane.
Draw a 20mmx20mm square with bottom left corner on the origin.
Pad it to a 20mm cube.
Now go back to the first body, make it active. Select second body and do a boolean. Select cut to see the issue. Boolean is done with body coordinates of 0,0,0, not 0,10,-10. The coordinates of the body are not included in the boolean calculations. 
If you reverse the process and make the moved body the tool, it seems to work as expected.

Roy-043, 2023-09-25:
Thanks for the clear steps and the file. Confirmed.

bgbsww, 2023-11-08:
This is interesting.  If I'm reading correctly, the code completely ignores the placement of the body.  If we apply the placement of the body to the incoming result shape before the boolean operation, we get the expected combination, but we now have an overall shape out of position by the placement value.

This could be corrected, but the overall patch is likely to break some existing files out there if the boolean gets recomputed.  Is this the kind of thing we should do?

https://github.com/FreeCAD/FreeCAD/blob/9be9881ed84247c1a1cde34b7cdad10d153f2489/src/Mod/PartDesign/App/FeatureBoolean.cpp#L107

bgbsww, 2023-11-08:
So, here's a potential fix.  It changes the behavior of PartDesign booleans though; they used to ignore the placement of the original body.  I think this means it could affect existing documents if the boolean is recomputed.

Should this progress?  If so, I'm happy to be assigned.

https://github.com/FreeCAD/FreeCAD/compare/main...bgbsww:FreeCAD:bgbsww-patch-5?expand=1

luzpaz, 2023-11-08:
@FlachyJoe would you have a moment to review ?

FlachyJoe, 2023-11-08:
@bgbsww Please open a draft PR to propose code in order to locate comments in the FreeCAD repository. And add a `Fix #10707` line in its comment to link it here.

@luzpaz please assign me as reviewer in the PR bgbsww will open.

bgbsww, 2023-11-10:
@FlachyJoe Agreed with your comment on #11321.  If there are historical documents that have Bodies with non origin placements inside Booleans then they are subject to breakage after recompute.  The question is really if that has happened in practice.  Not taking the placement into account seems like a historical oversight, but if we are stuck with it, so be it.

Maybe #10707 needs some discussion along with #11092 as to the overall desired outcome.  I'm a little nervous about clogging up the tree with Binders dropped into it, wouldn't that create a questions about visibility and where to place the tool objects with Binders after they are added to the Boolean?  Is the result better than the current state of affairs, or should we just leave it be?

bgbsww, 2023-11-14:
I'm not going to make more progress at the moment on this, as I'm uncomfortable with the proposed use of binders here.  I agree that the backward compatibility issue needs to be solved, but adding more complexity as opposed to correcting what appears to be an error in the past doesn't seem quite right to me.  So I'll keep thinking about it, but unassign myself.

bgbsww, 2023-11-14:
I spent some time looking for some project files that are using PartDesign boolean operations to see how many of them have non zero placements and thus would be negatively affected by simply making the placement work.

I was unable to find an examples of such a file, which probably means I'm looking in the wrong places.  Anyone know any locations of these files?

mwganson, 2023-11-16:
Related discussion

https://forum.freecad.org/viewtopic.php?t=82713

closed 2023-11-27 by yorikvanhavre