FreeCAD/FreeCAD #10708: [Feature request] New constraints to reduce flipping

PLChris, 2023-09-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

A set of constraints can describe multiple sketches, even if it is fully constrained; this is known as _flipping_.
I propose to add new constraints, "directed horizontal dimension" and "directed vertical dimension".

**directed horizontal dimension**
Applied to a geometric element, the direction is from left to right, starting at the start point of the geometric element going to the endpoint.
A positive value would go from left to right, a negative value would go from right to left.

Applied to different geometric elements, the startpoint would be the element with the lower index, the endpoint that with the higher index.

**directed vertical dimension**
This is equivalent with positive values going up and negative values going down.

### Full version info

```shell
[code]
OS: macOS 13.0
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34137 (Git)
Build type: Release
Branch: master
Hash: 7ca834acc58841dcf2d66d186db818c8f35f359d
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * FC_SU
  * freecad.gears 1.0.0
  * FeedsAndSpeeds 0.5.0
  * dxf-library
  * fcgear 1.0.0
  * fasteners 0.4.59
  * sheetmetal 0.2.62
  * ExplodedAssembly
  * Curves 0.6.8
[/code]
```

### Subproject(s) affected?

Sketcher

### Anything else?

[Forum discussion](https://forum.freecad.org/viewtopic.php?p=707208#p707208).

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

qewer33, 2023-09-17:
Instead of new constraints, imo it would be better to make the current horizontal/vertical distance constraints work with +/- prefixes. 

So for example when the value for an horizontal distance constraint is "+15" instead of just "15", the solver will make that constraint from left to right (in the +x direction if you think of the sketch plane as a coordinate system). And right to left if the value is "-15" (-x in the coordinate system). 

Normal values without +/- prefixes would work how they work currently, no changes there.

Pretty much exactly what is proposed here but making it work with the current commands instead of adding new commands.

Jolbas, 2024-05-11:
I think a better idea would be to add a direction lock option to all angular constraints (horisontal, vertical, angle, parallell and perpendicular and maybe tangential). The lock would weigh in the order of the endpoints in the involved lines so that if one line turns 180° the other line will turn too. I think this lock may even be bound to if it is construction geometry or not.

r-nd-m, 2025-08-31:
This feature is a must have. After updating to 1.0 I'm seeing a lot of flipped constraints in my migrated models - it's frustrating to a point that  I'm considering of abandoning FreeCAD if I would have to redo all my broken models every major revision, e.g. my PC I/O plate:
[image]
Is there any way to setup a bounty?

theo-vt, 2025-10-29:
That's right! I think that what is missing is for all cases where horizontal and vertical lines are used

1) A way to interpret dimensions as X/Y dimensions
2) A way to interpret equal length constraints as X/Y dimensions
3) A way to interpret perpendicular/parallel lines as horizontal/vertical