FreeCAD/FreeCAD #10709: [Problem] FreeCAD crashes when using polyline tool.

oscilococcinum, 2023-09-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD weekly-build crashes when i try to draw a polyline in sketcher.
I tried to recreate this problem on linux, but it seems its only windows issue.

https://github.com/FreeCAD/FreeCAD/assets/81778075/70caba95-e9b2-4ed5-9c32-2513c0431b79

### Full version info

```shell
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34291 (Git)
Build type: Release
Branch: master
Hash: 71f1e6b09baa45812026832dfd74e59b5d4d1d35
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly3 0.12.0
  * Curves 0.6.13
  * DesignSPHysics 0.6.3(24-09-2022)
  * Plot 2022.4.17
  * sheetmetal 0.2.63
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-09-17:
@FEA-eng if you're on windows, do you mind testing ?

FEA-eng, 2023-09-18:
I can't confirm this.

@oscilococcinum Is it repeatable or happened just once ? Can you try recreating the crash ?

[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34291 (Git)
Build type: Release
Branch: master
Hash: 71f1e6b09baa45812026832dfd74e59b5d4d1d35
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
[/code]

luzpaz, 2023-09-18:
@oscilococcinum can you also run it from the command line and see what the error generated is when it crashes ? Or start freecad with the logfile function ? 
Start FreeCAD with `--write-log` flag. Log file is in `%APPDATA%\FreeCAD`

oscilococcinum, 2023-09-19:
I will have access to my windows station tomorrow. I will send feedback as soon as I can.

oscilococcinum, 2023-09-20:
I generated log file.
[FreeCAD.log](https://github.com/FreeCAD/FreeCAD/files/12676103/FreeCAD.log)

luzpaz, 2023-09-21:
The log file is noisy due to 3rd party Mod code not getting initialized (3rd party addons missing InitGui.py). But very sparse about why it crashes.

xtemp09, 2023-09-21:
Well, @oscilococcinum, it seems you have to run FreeCAD within debugging environment (_lldb_, _gdb_ or _cdb_). I would recommend to download a portable toolchain at https://winlibs.com.

maxwxyz, 2023-12-18:
I cannot reproduce. @oscilococcinum could you try it again with the latest build?

maxwxyz, 2024-01-14:
@Roy-043 ^

Roy-043, 2024-01-14:
@maxwxyz Thanks for keeping tabs on things.

closed 2024-01-14 by Roy-043