FreeCAD/FreeCAD #10713: [Feature Request] Improved options for PartDesigns Draft and Fillet

cptshirty, 2023-09-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I find myself using PartDesigns Fillet and Chamfer feature quite often these days and I often get slowed down considerably by having to select **hard-to-reach edges**. I then realised, that if there was a way to select faces surrounding these edges and having the Fillet/Chamfer tool use only edges, that are surrounded by the Faces, instead of all edges Face has, it would improve my experience a lot.
I then implemented it on my fork of FreeCAD in a form of checkbox in the Fillet/Draft and found it incredibly useful, so I was wondering if it's worth making a pull request, so everybody has access to this feature without tinkering.

### Full version info

```shell
[code]
OS: Fedora Linux 38 (Workstation Edition) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34249 (Git)
Build type: Unknown
Branch: BetterFilletAndDraft
Hash: fd22cbd291a613992d6631bce20a4f2c8a4bbfdc
Python 3.11.4, Qt 5.15.10, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
[/code]
```

### Subproject(s) affected?

PartDesign

### Anything else?

# Images of the improved Fillet feature being used

note: The images below are only good for the purpose of clearly showing how this would work in UI, so I would like to reiterate that **the real benefit of this tool is for hard to reach edges**

## Select screen

[image]

## Preview screen

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

howie-j, 2023-09-17:
This sounds like a solution to solve a problem caused by an underlying issue, namely that some edges are difficult to select. There are at least two issues regarding that:

#9205 Problem: Default settings make it hard to select objects in the 3D window
#10308 [Problem] Selection order priority

In my opinion it's better to focus on fixing the main issue that edges are hard to select, than creating bandaid solutions.

cptshirty, 2023-09-18:
I would have to partially disagree with @howie-j. 
I would not consider this just a bandaid solution to the issues listed, although it can somehow help with those issues.

Even if we would ignore the selection order priority, selecting a face will always be easier than an edge.

Imagine this scenario: you would like to make a thin plate, that has square hole in the middle and you want to fillet the edges of the hole.
In the current workflow, if you don't want to switch your view to  wireframe (because that can be really confusing for some parts), you have to rotate the part, select edge, and repeat this 4 times.

With this feature, you can just rotate the part twice and select all the faces in the hole, which is much more pleasant in my opinion.

It could be argued,  that this is special case, but I would argue, that filleting all the edges on a part is much more special case than this.

howie-j, 2023-09-18:
I tried your example with the selection fix wmayer solved here: [issue 9205](https://github.com/FreeCAD/FreeCAD/issues/9205#issuecomment-1723236170)

I'm still not sure selecting the faces and then checking the option is much easier:

[issue_10713.webm](https://github.com/FreeCAD/FreeCAD/assets/82042580/5550fe47-eacd-4195-b3ad-30d566e290b0)

This is a 100x100x1 plate with a 10x10 cutout. Video is realtime.
Do you have a file or another example of hard to select edges?

cptshirty, 2023-09-19:
Thanks for the video!
The issue you are addressing was most prevalent with triangular holes, so you can try that, if you would like to.

I am unfortunately not able to make a video right now, but I tried to match you cursor speed and managed to do the same task about 30% faster, but that is not really that much of a difference, considering it takes about 7 seconds in the video. Additionally, another benefit of using the faces instead of edges is that when something goes wrong in the dependency tree, and fillet recomputes for different objects, than is is supposed to, spotting "rogue" face is much easier than edge (at least for me).

I think that the benefits, even if minor, are well worth it, considering that to implement this feature, I had to add like 10 lines of code (not including the gui stuff).

X-Ryl669, 2024-02-20:
Related: #12316

All in all, both express a user frustration with FreeCAD compared to other CAD software. This issue is focusing on simplifying the operations to perform, while my issue refers to avoiding some of these operations.

Typically, casual operations should be the fastest to perform and adding a fillet/chamfer is a very casual operation to perform. 
Usual way to specify a fillet is to select a closed path (and the software can compute the loop of edges by itself, why doesn't it do that by default?). The worst here, is that feature exists in FreeCAD but it's buried in another unrelated workbench thus inaccessible. 
Another way is to specify the edge(s) between two or more faces which are easier to select than a edge. 
The software knows that fillets and chamfers applies to edges, so if given a set of faces as input, it can completely figure out the intersection edges between them automatically.

maxwxyz, 2026-01-04:
FYI there is now a workaround with the Clarify Selection tool which also works in these cases where you have an open task to select something. Either long press the left mouse button over the area of interest or right click and choose Clarify Selection. Hovering over the entries then highlights the edges, points, faces or objects and a click confirms it.