FreeCAD/FreeCAD #10717: [Problem]  Follow up to https://github.com/FreeCAD/FreeCAD/pull/10552 in Preferences>Display>Colors: Tree view and Color Bar changes do not update current objects on OK Clicked

Syres916, 2023-09-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Only when new objects are created are the color and text size settings used based on Preferences>Display>Colors for _Tree view_ and _Color Bar_ 
Existing objects would ideally update when the user clicks OK to complete the Preferences window.

### Full version info

```shell
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34317 (Git)
Build type: Release
Branch: master
Hash: beb39239f394888e9bd47df3b18456b6f8aa1131
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 

with https://github.com/FreeCAD/FreeCAD/pull/10552 applied
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-10-09:
Any traction on this ?

Syres916, 2023-10-10:
@0penBrain is aware as at present it's beyond my skill set, it's not super urgent but would be nice to get it sorted by Xmas.

0penBrain, 2023-10-10:
> t's not super urgent but would be

I'm lacking time for this now. But if you want to give it a go, I can point you out to somewhere in the code where something similar is implemented. ;)

luzpaz, 2023-10-24:
@0penBrain can you offer those insights in this ticket when you get a chance ?

0penBrain, 2023-10-27:
> @0penBrain can you offer those insights in this ticket when you get a chance ?

Well, basic principle is as follow.

In the ctor, ViewProvider shall attach to the appropriate parameter group as here: https://github.com/FreeCAD/FreeCAD/blob/7bac7956a49c4759e4be92a7240c5c020a74b488/src/Mod/Sketcher/Gui/TaskSketcherConstraints.cpp#L863

Then it has to implement an `OnChange` method. It will be called each time a parameter is changed in the group: https://github.com/FreeCAD/FreeCAD/blob/7bac7956a49c4759e4be92a7240c5c020a74b488/src/Mod/Sketcher/Gui/TaskSketcherConstraints.cpp#L1408

Think about detaching in the dtor: https://github.com/FreeCAD/FreeCAD/blob/7bac7956a49c4759e4be92a7240c5c020a74b488/src/Mod/Sketcher/Gui/TaskSketcherConstraints.cpp#L976

Syres916, 2023-12-28:
As I expected, this is way beyond my capabilities, just getting more frustrated with every new error so abandoning it for someone with more C++ capabilities.

0penBrain, 2024-04-27:
@Syres916 I had a look at this and well ...
 * For Tree View, the code is so convoluted that this would be a PITA to implement. Not enough spare time on my side to to do this
 * For Color Bar, the problem is that it is used on several places and relies on the implementation of a custom Coin node. The "simplest" solution would be that this custom node attaches to FC preferences, but I'm not sure how to do that properly, or even if it's a good idea to do so. Maybe @wwmayer can provide some insights on this.

closed 2025-02-14 by github-actions[bot]