FreeCAD/FreeCAD #10718: [Sketcher] Colinear constraint

dyylanhammond, 2023-09-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The tangent constraint currently serves as the colinear constraint for freecad. While this is fine in implementation, it isn't good for discoverability since tangent and colinear aren't usually synonyms.

I see 2 possible solutions:
Update name, tooltip, and possibly icon to reflect that it's a dual purpose constraint
Or
Separate tangent into tangent and colinear.

### Full version info

```shell
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * PieMenu 1.2.7
```

### Subproject(s) affected?

Sketcher

### Anything else?

See also #10704

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2023-09-18:
> I don't see any issue with that name, but making an icon is difficult.

Me neither but it might be best to discuss this on the forum first.

>but making an icon is difficult.

Maybe @PaddleStroke could suggest something.

PaddleStroke, 2023-09-18:
I agree that its counter intuitive. I spent a long time not knowing that tangent can do colinear.
Separate constraints could be the best solution but this adds one more icon which is not ideal. But yes at least if the name/tooltip was clear it would go a long way.

For the icon I don't have great ideas. Yours seems acceptable.

closed 2024-03-04 by yorikvanhavre