FreeCAD/FreeCAD #10719: Transform dragger remaining issues from #10706

luzpaz, 2023-09-18:
> > any chance you can take a look at https://github.com/FreeCAD/FreeCAD/pull/10706#issuecomment-1722431251

> The PR has been merged before I had the chance to look at the code. Basically the code looks good and seems to work as expected but there are a few issues that should be solved:
> * Several empty newlines between two functions are removed. They should be kept.
> * SoFCCSysDragger has a dependency to  the singleton ViewParams now which has a code smell because it has global behaviour. It's actually as bad as working with global variables. Instead you can add some public class members of type SbColor to SoFCCSysDragger and then set them inside ViewProviderDragger (or TaskCSysDragger)
> * In SoFCCSysDragger don't use App::Color but SbColor

_Originally posted by @wwmayer in https://github.com/FreeCAD/FreeCAD/issues/10706#issuecomment-1724082967_

qewer33, 2023-09-18:
I'll look into these tomorrow and open a following PR to fix them.

qewer33, 2023-09-21:
@luzpaz this can be closed

luzpaz, 2023-09-21:
@qewer33 JFYI, https://docs.github.com/en/issues/tracking-your-work-with-issues/linking-a-pull-request-to-an-issue#linking-a-pull-request-to-an-issue-using-a-keyword

closed 2023-09-21 by luzpaz