FreeCAD/FreeCAD #10724: [Problem] New TaskView gets truncated

PaddleStroke, 2023-09-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The new taskview gets truncated if it's too small for current task. It does not resize automatically as it was doing before. 
@wwmayer
[image]

Also the stylesheets seem broken by the changes. @MisterMakerNL

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34337 +3 (Git)
Build type: Release
Branch: Assembly_joint_basis
Hash: 4c01add58ec2209d9b9ed3a5883c9ab92718a6f7
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk , OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.60k
  * Assembly3 0.12.0
  * Assembly4 0.50.2
  * Curves 0.6.8
  * ExtremeProLight 1.5.0
  * FreeCAD - Shortcut.lnk
  * FreeCAD_FastenersWB-master 0.4.56
  * Ondsel 2023.8.23.02
  * ProDarkThemePreferencePack 1.0.0
  * sheetmetal 0.2.63
[/code]
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MisterMakerNL, 2023-09-19:
Seems like the width is determent by the buttons and not the Qsint widget completely.
[image]

PaddleStroke, 2023-09-19:
Sorry! My mistake. Somehow I never noticed it before. I edited the issue then.

MisterMakerNL, 2023-09-19:
I assume it's not easy or even possible to maje this dynamic? If so, we might need to make design rules to the minimum width?

wwmayer, 2023-09-20:
The grand-grand-parent widget of a task dialog is a QScrollArea and setting a minimum width solves the problem. 

If you open the command to create primitives in the PD workbench then test this code:
```
td = Gui.Control.activeTaskDialog()
# it contains two widgets
w1,w2 = td.getDialogContent()
# frame is a QFrame
frame = w2.parentWidget()
layout = frame.layout()
# this is the view port widget of the QScrollArea
viewport = frame.parentWidget()
scroll = viewport.parentWidget()
scroll.setMinimumWidth(w2.width() + 4 * layout.spacing())
```

PaddleStroke, 2023-09-20:
So maybe this code should be called everytime the taskdialog gets or looses a new task box? This way it would resize automatically to the correct size.
Cherry on top it could resize to 0 when no taskbox is present. This way the panel would hide when not needed.

PaddleStroke, 2023-09-20:
Yes, when I said this code I meant didn't meant this python snippet.

wwmayer, 2023-09-20:
Of course.
https://github.com/FreeCAD/FreeCAD/pull/10762

commit fdc44d50 (2023-09-22): Gui: fixes #10724: TaskView gets truncated
commit 239f5a0d (2023-11-10): Gui: fixes #10724: TaskView gets truncated
closed 2023-09-22 by chennes