FreeCAD/FreeCAD #10725: [Problem] FreeCAD has 2 report views.

PaddleStroke, 2023-09-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Recently @abdullahtahiriyo introduced the 'non intrusive notifications' which is a great idea.
However the implementation also included an additional button and view, such that we now have 2 different report views.

[image]

I think that this is wrong. There should not be 2 competing solutions doing the same thing. It adds needless complexity, confusion and clutter. For no visible gain that I can think of.

If one has functionality that the other doesn't have, just bring these functionalities to the existing view.

Proposal : 
- We remove the 'notification area' view.
- We keep the new 'message' button.
- When the message button is clicked, it toggles the visibility of the report view.

### Full version info

```shell
0.22 dev
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

abdullahtahiriyo, 2023-10-03:
The notification area is intended for the user and there is a requirement that messages appearing there are translated (unless explicitly configured in settings to show developer messages too). 

The report view was never intended for end users, but for developers (or at least power users). Developer messages will only appear there. Messages in the report view are untranslated (in English). It is intended that regular users do not have the report view open.

The notifications area does not substitute the traditional report view.

Developers may want/need to see developer messages in the notification area. There is a setting in preferences for that. This setting is not intended to be active for end users.

Error/Warning functions of Console() now have several variants, so that the developer can control exactly what appears where.  This was done to end the controversy about users complaining of being shown "developer messages" and to promote that errors facing the end user are meaningful (which are not). So you can have DeveloperError and UserError, which define where the message will end up. There is also TranslatedUserError for those cases where the developer provides an already translated message intended for the notification area.

I want to write a flowchart about how to select the right function from all these, and also the Notification framework that is available to substitute pop-up messages with user notifications painlessly. I need to have some more free time (and access to the wiki). Hopefully it will happen over this October or November...

PaddleStroke, 2023-10-04:
I see your point. However right now the default setting is 'show report view on error'. And errors are clearly not only intended for devs right now.
So the console is in fact currently a user facing view.
On top of that users are used to this view. But habits should not be a limiting factor I guess.

But anyway my previous argument are mostly only temporary I guess.

But still I think having 2 report views is still a bad move at the end of the day. The report view could be modified to work properly rather than introducing another view.

abdullahtahiriyo, 2023-10-04:
I am really not sure you are getting my point.

What I read from your last line is: Remove the notification area and improve the report view.

You cannot "modify" the report view to turn it into what we want for the notification area without destroying what we want for the report view.

The whole point of the notification area is to do away with the report view for end users and avoiding modal dialogs for notifications, while ensuring messages intended for the end user are curated and translated and ensuring that the report view is kept in English, so that developers can understand logs and errors across languages, while user messages should appear in the UI language.

The report view writes the log. The notification area ignores logs. The report view ignores user notifications (like the one that you cannot add a vertical constraint because the element already has one). The notification area shows it.

The two are intended to cover different functionality.

The (python) console does not appear to have anything to do with the report view and the notification area.

Users who do not use the python console want to have all screen space for the 3D view. They do not want the report view.

Power users that want to have the python console can have more space in it by closing the report view. If they need developer messages, they may want to redirect the developer errors to the notification area. Of course, if they want, they can keep the report view too. If they do not want the notification area, they can disable it.

There are errors and warnings that are not intended for the user and currently appear in the notification area (when redirection of developer messages is not intended). This is because the error messages are not curated yet.

In the medium term we should:
- Curate the error messages intended for the user.
- Make "show report view on error" setting disabled by default.

That way, the user will have the information he needs in the notification area. He won't see errors intended for developers. He will see everything in his UI language.

yorikvanhavre, 2023-10-09:
I agree both are needed but should not "overlap". I would say we should:

1. Treat the report window more as for power-users developers. A bit like ubuntu has become  progressively usable without ever touching the terminal, but the terminal still a fundamental tool for anything more advanced
2. Having maybe a `FreeCAD.Console.PrintNotification()` that could maybe be automatically picked up for translations? Not sure how to achieve that :)
3. Create a clear guide to tell devs what should go to PrintMessage and what should go to PrintNotification
4. If the report view is open, print notifications there instead of the notif area (or have an option to define where to print notifications)

abdullahtahiriyo, 2023-10-13:
I added a chart about the c++ messages:
https://forum.freecad.org/viewtopic.php?t=81809

[image]

1 => Agree.

2 => From Python, I do not know how to add translations to the QT translation engine dynamically during execution (the equivalent of QT_TRANSLATE_NOOP, but adding them directly during runtime). This must be possible and probably you know how to do it already.

If one adds them to a context "Notifications", then there is already a Framework in place for Python that will translate the message using the QT framework. 

When the message faces the user (it is not intended for the developer only), the Notifications Area will always try to provide a translation for the message.

For those translations currently in FreeCAD, that have the "Notifications" context, FreeCAD will try to translate it. For example, (select your favourite non-English language in FreeCAD UI and do):

```FreeCAD.Console.PrintError("The Sketch has malformed constraints!")```

As shown in the chart, this one is intended to sending an error message to the user and to the developer. The result is that in the report view you will get the original English error. However, in the notifications area, you will get the translated version. I do get: "Die Skizze enthält fehlerhafte Randbedingungen!"

As it is the case with C++, you can do PrintUserError, when the message should only be presented to the user, and PrintDeveloperError when it should only be presented to the developer (the first will only appear in the Notifications Area and the second only in the report view). Of course, you can substitute Error with Warning (for severity warning) or send a Notification (which is always intended for the user, and not the developer, as shown in the chart, for severity information).

3 => Ok. 

The chart does not consider the legacy messages that are unsupported by the Notifications Area, such as PrintMessage. That one does exactly what it did before and the NotificationsArea will not show it. 

The NotificationsArea only handles the messages indicated in the chart. So no "message" or "log".  

I hope the chart is clear enough. If not, let me know. I still have the plantuml source code and can amend it.

4=> I am not sure this is sound in those exact terms.

There is in place the opposite option: Redirect the developer messages to the Notification's Area, with the developer messages appearing in English there if no translation is provided in the framework. So that the report view can be kept closed (available from settings).

If a developer wants to see only the report view, this is also possible, as the Notification's Area can be disabled. However, he will not receive messages of type Notification (severity information) in the report view, or those for which the already translated message is provided. This decision is deliberate, so as not to pollute the report view with localised strings that would make little sense for developers (We talked about getting some nice Japanese strings in the middle of the Report view, and not being able to help the user. With the English version we can, even in the Japanese user is not that good at English).

My guess is that a developer will only want to have the report view exclusively in exceptional circumstances, the reason being that a developer generally wants to see what the user will receive as notifications, as it is part of the development process. It will even become more the case when we actively seek to provide curated messages only to the user, while hiding those that are only for developers in the report view.

Of course, if you think we still need something else, we can talk about the specific use case.

MisterMakerNL, 2023-10-20:
Why make is the reportview a panel if it isn't for normal users? Kick it out of normal users ui move it to a seperate window.
Power users have more then one screen. And separating it from normal ui makes it clear it doesn't belong there. It simplify things and makes it more uniform.

abdullahtahiriyo, 2023-10-21:
For normal users it should be hidden. Even for developers, often it is better to have it hidden (it depends on what one is working). In other words, during this development cycle it should be set, by default, hidden.

When it is not hidden, it can be undocked (made floating). I do not know if it can be moved to a different screen (I only have one screen). If it is floating, it can be moved within a screen outside the area used by FreeCAD or behind it. I do not find much use for this, but it can be done.

X-Ryl669, 2024-02-02:
Would it be possible to redirect the report view to stderr (so it's visible on the terminal that launched FreeCAD and dropped if launched from the desktop/GUI). It's very likely that the information in the report view is only useful for some dev only when (s)he is developing, thus launching the app from the command line. I'd say that most of the time, you don't actually care about the report view and it's wasting screen estate for nothing. 

When something goes wrong (an operation that can't be performed or whatever), the report view that pops up is frustrating and not helpful at all, since there is no limit to show where the last "error" message stops. It'll have collected maybe hundreds of previous "failure" or warning or errors and since there's no ">" or prompt like, you can't be sure that an error message is related to the last operation that failed. In the end, I find it completely useless.

So maybe while modifying this it would be interesting to change the function interface that's dumping in the report view to include a "Operation Name" and "Operation index" parameter so that operations could be folded/merged, or at least, identified in the view. 
Even better, being able to redirect the output to stderr by default and avoiding that pesky pane would be better.

JohnOCFII, 2025-11-18:
> In AstoCAD I changed the notification area button to toggle visibility of consoles :

That definitely is an improvement.

maxwxyz, 2026-02-07:
From the 2026 complaint session:
**There are two different ways to report failures and they are inconsistent (report area versus notification)**