FreeCAD/FreeCAD #10728: [Problem] Missing API documentation for `src/Mod/Part/App/HLRBRep`

luzpaz, 2023-09-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently the https://github.com/FreeCAD/FreeCAD/tree/master/src/Mod/Part/App/HLRBRep XML files show no documentation about said HLRBRep functionality. Discovered in #10726 & #10727

https://github.com/FreeCAD/FreeCAD/blob/4293c4a65b0e69975dcdbbf68c80d8019a4f850d/src/Mod/Part/App/HLRBRep/HLRBRep_AlgoPy.xml#L15-L106

https://github.com/FreeCAD/FreeCAD/blob/4293c4a65b0e69975dcdbbf68c80d8019a4f850d/src/Mod/Part/App/HLRBRep/HLRBRep_PolyAlgoPy.xml#L15-L114

https://github.com/FreeCAD/FreeCAD/blob/4293c4a65b0e69975dcdbbf68c80d8019a4f850d/src/Mod/Part/App/HLRBRep/HLRToShapePy.xml#L15-L87

https://github.com/FreeCAD/FreeCAD/blob/4293c4a65b0e69975dcdbbf68c80d8019a4f850d/src/Mod/Part/App/HLRBRep/PolyHLRToShapePy.xml#L15-L81

### Full version info

```shell
v0.22.x master branch
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pieterhijma, 2023-10-04:
I haven't had time yet, Is it time critical? If so, I could increase its priority, if you prefer.

luzpaz, 2023-10-06:
Not critical.

RohanMishra315, 2023-10-28:
Hey @luzpaz should I work on it. Already make the changes just need to do PR.

luzpaz, 2023-10-28:
@RohanMishra315 yes please. Thanks!

pieterhijma, 2023-11-06:
@RohanMishra315, great that you could work on it! Looks like a great start, but there are other files that lack documentation as well as @luzpaz mentioned above. Next week I will have time to take a look as well, if you want, we can work on it together. Otherwise, I will build on your changes.

PR #11478 (2023-11-22): [Part] Missing API documentaton Python HLR
closes #10728.

Note, that this PR overrides  #11233.

It is based on the same commit of @RohanMishra315 but builds on it.

This PR makes the following contributions:
- Since OCCT has a C++ interface, the Python interface needs to be adapted. I carefully checked the types of the Python bindings and matched the documentation accordingly.
- OCCT has lots of missing documentation. I tried to fill in missing pieces as much as possible. As a result, the HLR documentation in FreeCAD's Python interface is now better than OCCT's! (As an example [rg1LineVCompo [... cut]
closed 2023-11-22 by chennes